introlab/odas #2: Any difference between odas and manyears?

zhaoforever, 2017-12-14:
Hi  Francois,
      I have used the manyears project and it works well, and the papers you provided helps a lot. As for the odas project, is there any difference between the odas and manyears? If so, would provide some papers for reference? Thanks.

FrancoisGrondin, 2017-12-21:
Yes many differences. ODAS is meant to be faster, more robust to various microphone array shapes, and in some cases improve sound source separation. Papers are on the way, and documentation is currently being written in the wiki. A demo video will be released soon. Stay tuned!

zhaoforever, 2018-01-24:
@FrancoisGrondin Thanks for the demo video. The noise cancellation and sound source separation results seem great! Looking forward to the wiki.

closed 2018-01-24 by FrancoisGrondin