introlab/odas #12: Respeaker.cfg segmentation fault

taospartan, 2018-03-14:
Is this config for the respeaker mic array? The config throws asegmentation fault when used

FrancoisGrondin, 2018-03-14:
Hi,

This has been tested and should work. However, from what I remember, the respeaker may have a different firmware by default, which prevents us from using it as a 7 inputs usb sound card. Have you updated the firmware on the Respeaker to use it as a sound card?

Thanks,

Frank

taospartan, 2018-03-14:
Thanks for the quick reply, yes it’s got the 8 channel raw firmware. I’ll run some more tests tomorrow on the mic array tomorrow to check if the card itself is faulty.

FrancoisGrondin, 2018-03-14:
Ok, I suppose you have tested the sound card with a software such as Audacity? If so, you could record the raw signals from your sound card (8 channels, 16-bit signed, pcm, little endians, 16kSamples/sec), and send it to me along with the exact .cfg you are using. If I can to reproduce the bug on my side, I should be able to find what's wrong.

taospartan, 2018-03-15:
Unfortunately the problem wasnt solved, still a segmentation error. Here is the arecord -l

**** List of CAPTURE Hardware Devices ****
card 1: Array [ReSpeaker Microphone Array], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Here is the error:

+--------------------------------------------+
|    ODAS (Open embeddeD Audition System)    |
+--------------------------------------------+
| Author:  Francois Grondin                  |
| Email:   francois.grondin2@usherbrooke.ca  |
| Website: introlab.3it.usherbrooke.ca       |
| Version: 1.0                               |
+--------------------------------------------+
| + Initializing configurations...... [Done] |
| + Initializing objects............. Segmentation fault

[reSpeakerArray copy.cfg.txt](https://github.com/introlab/odas/files/1816045/reSpeakerArray.copy.cfg.txt)

FrancoisGrondin, 2018-03-15:
It seems the configuration files were not fully updated to match the latest version of ODAS. It did update them, it is now in the master branch. Please update your respeaker.cfg according to the new version, and let me know if this fixes the issue.

taospartan, 2018-03-15:
Yes success! it works now many thanks!

closed 2018-03-15 by FrancoisGrondin