introlab/odas #13: beam-forming

taospartan, 2018-03-15:
Im interested in focusing the microphone array to a specific point in space ( as that is where the sound will always come from) whilst cancelling the background noise. Is this possible within the scope of odas? Is it configured with the config file?

FrancoisGrondin, 2018-03-15:
Yes you can do that, with what is called "target sources". In the cfg file, change the following in the "sst" block:

If you only want this source to be tracked:
add = "static";

You can leave this to dynamic if you also want the system to automatically detect new sources different from the target one.

And then replace "target: ();" by:

target = (
        { tag = "myTrackedSource"; x = <x-coordinate>; y = <y-coordinate>; z = <z-coordinate> }
);

You can choose the tag label you want (this tag will propagate in the tracked sources to make the association easy). The x-, y-, and z-coordinates should be normalized (x^2+y^2+z^2 = 1): remember your are dealing with a direction in the far field, so the norm of the vector is always normalized to 1.

taospartan, 2018-03-15:
Many thanks, that’s great!

closed 2018-03-15 by FrancoisGrondin

jamesshao8, 2018-05-17:
I tried the modification in cfg file to only record the sound from the y axis, but there is no audio file recorded.

I did my modification according to your suggestion:

add = "static"; 

target = (
{ tag = "myTrackedSource"; x = 0; y = 1; z = 0}
);

Once I changed the modification back to original, I am able to record, so I think the software and the hardware is still working.

Could you please take a look? Thanks.

jamesshao8, 2018-07-09:
@taospartan Hello I am having the same issue as you had. But I still can't record from a fixed direction. Could you please provide me your cfg file? And are you using respeaker v2.0 instead of v1.0? Thanks.

jamesshao8, 2018-07-09:
@taospartan Thanks. I will test it with audacity. But for now, if I use respeaker v1.0 and v2.0 under dynamic mode, both of them can successfully record separated wav files normally. Only when I use static mode, I have no file recorded. Could you please share with me your configuration file? So I can try it here with respeaker v2.0? Thanks again.

taospartan, 2018-07-09:
Are you using odas_web? If so that’s the reason. Write the sep and pf to type = file. I don’t have the Cfg with me at the moment.

Import the outputed raw file to audacity ( remember it uses 4 channels) play the first channel and you should hear  the beamformed audio. I think odas web is waiting for the audio source to disappear which it won’t do as it’s static, before it writes a file. That’s why it works under dynamic. 

Hope this helps

taospartan, 2018-07-09:
> @taospartan
>
> This is our respeaker v2.0 working in audacity. It looks strange as the signal strength of channel 1 is much higher than other 3.
>
> We are trying both in odas_web and our own program (connected with odascore through socket). Our setting for sep and pf is now type=socket. And we found in our case, if we want to record wav files, we will have to use type=socket. Otherwise odas_web would record empty files.
>
> Inspired by your idea, we found when we initially use dynamic in cfg file and switch it to static, then we are able to record under static.
>
> But the recorded file has only 1 channel when opened by audacity.
>
> We will then try to use type=file in cfg and see if we could replicate your experiment.
>
> By the way, have you tried the 16SoundUSB? Theoretically more channel would have better spatial resolution. We have made one based on the design of introlab. Will see if it will produce better result.

When recording using audacity the first channel of the ReSpeaker 2.0 is the output from the processed audio by the ReSpeaker itself. Set the number of channels to 6 in audacity. From the ReSpeaker 2 wiki

Channel 0: processed audio for ASR 
Channel 1: mic1 raw data 
Channel 2: mic2 raw data 
Channel 3: mic3 raw data 
Channel 4: mic4 raw data 
Channel 5: merged playback

Ignore the 1st and 6th channel. 

Now I think the ReSpeaker 2 array config in odas is wrong as if you look at the mic coordinates in the cfg they describe a straight line array. I would double check yours.

I have only used a minidsp Uma-16 mic array but not in any depth, I’m still playing:)

jamesshao8, 2018-07-09:
[respeaker_4ch_fix_dir.txt](https://github.com/introlab/odas/files/2175637/respeaker_4ch_fix_dir.txt)
Thanks. This is my cfg file for respeaker v2.0.

taospartan, 2018-07-09:
try changing your mics to this

 mics = (

        # Microphone 2
        {
            mu = ( +0.0239, +0.0239, +0.000 );
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000,
+0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 90.0, 100.0 );
        },

        # Microphone 3
        {
            mu = ( -0.0239, +0.0239, +0.000 );
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000,
+0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 90.0, 100.0 );
        },

        # Microphone 4
        {
            mu = ( -0.0239, -0.0239, +0.000 );
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000,
+0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 90.0, 100.0 );
        },

        # Microphone 5
        {
            mu = ( +0.0239, -0.0239, +0.000 );
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000,
+0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 90.0, 100.0 );
        }

    );

jamesshao8, 2018-07-09:
Great. I will try that tomorrow.

jamesshao8, 2018-07-10:
@taospartan I tried your cfg today, the result is the same. I can always do DOA, but when recording in static mode, there is no beamforming effect.

https://github.com/respeaker/usb_4_mic_array/blob/master/odas.cfg

According to the cfg file from respeaker, I think their coordinates describe also a circular array. It's just that their axis is rotated from yours with an 45 degree angle. And it's also scaled with a factor of 10.

taospartan, 2018-07-10:
Can you post your config you used so I can have a look?

jamesshao8, 2018-07-10:
In my cfg file, I am using dds and ms. I will change the ms to to ss. Do you think I need setup some specific parameters into the field of dds?

taospartan, 2018-07-10:
> Did you perform beamforming using odas_web + odas_live? or just use odas_live in a standalone mode? We found that odas_web seems to have bug, we have to start recording in dynamic mode first, and then switch to static mode, not sure if it's the same at your place.

I just use odaslive

taospartan, 2018-07-10:
Just a thought is the beamforming out by 45 degrees?

taospartan, 2018-07-10:
Also I have tried netcating the separated audio into aplay that worked for me, odas web is a bit heavy as I’m on embedded devices. 

I can only think your issues are with odas_web

jamesshao8, 2018-07-10:
I will try to use odas_live without odas_web and try different possibilities. Thanks.

jamesshao8, 2018-07-13:
@taospartan I am now able to see beamforming effect in static mode, and I am using odas_web. Thanks for your help. Now I am eager to see if I can get better performance ( higher resolution or longer distance) with a 16 channel mic-array. I am still not able to get my home made 16SoundsUSB working. How's your minidsp Uma-16 going?

taospartan, 2018-07-13:
From initial testing ( I wouldn’t put too much faith in my work:) ) I’ve found that using the Uma-16 in a horizontal configuration was allowing me to beam form to sound sources at 20m plus. I found this counter intuitive as I expected if the array was vertical the results should have been better than horizontal but it wasn’t the case, I’m trying to work this out but have not conclusively reached an answer as yet. Do you have any ideas?

 I really like the Uma-16 but it does require a 12v dc power source as well as the usb connection which makes things awkward.the recording capabilities are outstanding.

Tao

11721206, 2019-06-28:
@taospartan jamesshao8 hello. can you give some tip for beamforming based on oda？and show the result after for beamforming. thanks

11721206, 2019-06-28:
@taospartan @jamesshao8 hello. can you give some tip for beamforming based on oda？and show the result after for beamforming. thanks

ruelj2, 2021-04-09:
> @taospartan I am now able to see beamforming effect in static mode, and I am using odas_web. Thanks for your help. Now I am eager to see if I can get better performance ( higher resolution or longer [...]

Can you please explain how you got it to work. On my side, I still have only 1 channel at the output of separated and beamforming doesn't seem to work at all. I've tried dds and dgss with static (x-0, y=0, z=1).

ruelj2, 2021-04-09:
Got it to work but I had to use app="dynamic" with tags. The postfiltering outputs 4 channels listing the tags before the dynamics.
Here is my config for the usb 4 mic array respeaker and an mp3 sample.

I'm using Gstreamer to record the output channels:
```
gst-launch-1.0 tcpserversrc host="127.0.0.1" port=9004 ! "audio/x-raw,format=S16LE,rate=16000,channels=4,channel-mask=(bitmask)0xf" ! audioconvert ! wavenc ! filesink location="output.wav"
```

[beamforming_usb_4_mic_array.mp3.txt](https://github.com/introlab/odas/files/6283727/beamforming_usb_4_mic_array.mp3.txt)
[respeaker_usb_4_mic_array.cfg.txt](https://github.com/introlab/odas/files/6283728/respeaker_usb_4_mic_array.cfg.txt)