introlab/odas #14: beamforming range advice

taospartan, 2018-03-18:
Can any settings be adjusted to increase the ‘range’ of the source detection? I’m looking at using odas for long range wildlife sound capture, using a static source tracking with a vector perpendicular to the array. 

Many thanks

FrancoisGrondin, 2018-03-18:
Hi,

I'm not sure I understand what you mean exactly by "range" of the source detection? Can you be more specific?

Thank you,

taospartan, 2018-03-18:
[image]

Please see the diagram. I'm interested in 'tuning' the mic array to differing distances in order to record wildlife at ranges above 10m. By using the beam-forming of odas, is it possible to 'focus' the mic array at a distance of 20m or 30m as shown in the diagram. For example increasing the amplification of the sound from the source or cancelling more background noise. The mic array will always be set to the static vector as shown, perpendicular to the mic array, the user would just have to point the mic array in the direction of the sound source.

hope this makes things clearer:)

FrancoisGrondin, 2018-03-19:
Thank you for the clear explanations and the figure :) Unfortunately, you cannot get the distance or focus the beamformer for a specific range if your sensors are closed to each other compared to the target source. I have written a small explanation in the new FAQ section as this question comes back very often. Please have a look here: https://github.com/introlab/odas/wiki/Frequently-Asked-Questions

I hope this answers your questions,

Thank you

FrancoisGrondin, 2018-03-19:
The `injector` module is designed to use either a target for a constant direction, or can refresh the directions over time (for instance you could send the new directions via a socket). The latter feature has not been implemented yet, but the architecture allows this to be done quite easily. If this is something you find useful, I'll add this feature asap.

As for your current issue with blank files for separated and postfiltered signals, I will have to look at your configuration (.cfg). Moreover, can you record the multichannels audio stream in raw format (16-bit signed little endian). With both the cfg and raw files, and can reproduce your issue on my side and troubleshoot.

taospartan, 2018-03-19:
That would be great feature to implement, the ability to 'steer' the beam possibly using odas_web would be a brilliant feature. 

I've done some investigation with the raw files and found that they are actually being recorded at 8000 rather than 16000 than specified in my config.

Do the raw files contain the number of channels that are given by the mic array? 

I've attached the cfg.

many thanks for devoting so much time to this.
[ReSpeaker.cfg.txt](https://github.com/introlab/odas/files/1827023/ReSpeaker.cfg.txt)

FrancoisGrondin, 2018-03-19:
Your config file seems good. 

So let's not be confused here: there are RAW files generated by separation and postfiltering, but here I'm referring to the RAW file obtained directly from the microphone array, before this is processed by ODAS. In this case, if your soundcard provides 8 channels (even though you select only 7 of them in the `mapping` section), your RAW file should contains the samples in the following structure:

`Channel 1, Sample 1, in 16-bit signed short (little endian)
Channel 2, Sample 1, in 16-bit signed short (little endian)
Channel 3, Sample 1, in 16-bit signed short (little endian)
Channel 4, Sample 1, in 16-bit signed short (little endian)
Channel 5, Sample 1, in 16-bit signed short (little endian)
Channel 6, Sample 1, in 16-bit signed short (little endian)
Channel 7, Sample 1, in 16-bit signed short (little endian)
Channel 8, Sample 1, in 16-bit signed short (little endian)
Channel 1, Sample 2, in 16-bit signed short (little endian)
Channel 2, Sample 2, in 16-bit signed short (little endian)
Channel 3, Sample 2, in 16-bit signed short (little endian)
Channel 4, Sample 2, in 16-bit signed short (little endian)
Channel 5, Sample 2, in 16-bit signed short (little endian)
Channel 6, Sample 2, in 16-bit signed short (little endian)
Channel 7, Sample 2, in 16-bit signed short (little endian)
Channel 8, Sample 2, in 16-bit signed short (little endian)
...`

If you could send me this file (obtained directly from the recording), I could have a look. For now ODAS can read this file directly (instead of reading from a soundcard) and generate outputs, so this allows you to reproduce results once the input is recorded and stored in a .raw file. I haven't add yet the possibility to record the raw from the soundcard directly while ODAS is running (e.g. you localize/track/separate and generate separated and postfiltered streams in files, and you also store the raw input in another file). For now you can record this file with Audacity or similar free software.

taospartan, 2018-03-20:
Here is a raw file recorded at 32Khz 16bit little endian using arecord and the ReSpeaker array
[audio-8ch-32k-14m.raw.zip](https://github.com/introlab/odas/files/1828748/audio-8ch-32k-14m.raw.zip)

The sound source is 14m away from the array.

FrancoisGrondin, 2018-03-20:
Ok so I ran the code with your RAW file. I get both separated and postfiltered streams, and there is something in the stream. I have joined the RAW files for separated/postfiltered, and the .cfg file. Don't forget that these RAW files are four channels because the maximum number of simultaneous tracked sources is set to four. To hear the results, open the raw file as 16-bit signed little endian with four channels, and play the wave in the first channel.

Please let me know if this works,

[audio-8ch-32k-14m_sep-4ch-16k_pf-4ch-16k.zip](https://github.com/introlab/odas/files/1830937/audio-8ch-32k-14m_sep-4ch-16k_pf-4ch-16k.zip)

taospartan, 2018-03-20:
the files do play:) I have gone back and carried out another test from the sound card. I am now getting the separated and post filtered raw files but even though the output raw file should be set at 16000Khz   in the odas cfg, when i import them using audacity, the sound files are actually 8000Khz. Is there any reason for this? 

Apart from that the issue has been resolved! Many thanks for the time spent on this with me

taospartan, 2018-03-21:
> Strange. It seems fine when I import them using audacity on my side. Are you using these settings: Signed 16 bit PCM, little endian, 4 channels, sample rate = 16000?

Yes, those are the settings on audacity

FrancoisGrondin, 2018-03-21:
Ok thank you. I'll keep an eye on that to see if I am missed something.

closed 2018-03-21 by FrancoisGrondin