introlab/odas #18: Support linear microphone arrays

skeggse, 2018-03-20:
It'd be nice to have support for linear microphone arrays, with tracking on a 2D arc instead of a 3D sphere.

FrancoisGrondin, 2018-03-23:
I'm working on this during the weekend. I thought about how to do this... and I'm tempted to allow the user to specify many spatial filters in the config file. The search space will consist of the union of these spatial filters. In this way, we could "squeeze" the search space to a donut-like shape (with two spatial filter pointing in opposite directions) when searching on a circle only, and we could add a third spatial filter to enforce search on a given arc. This way we always start with a full sphere, and shrink it to the desired subspace. Seems like an elegant solution, with minimal code refactoring. Plus it allows the user to add other spatial filters to deal with specific reflections coming from surfaces in the environment.

I'll keep you updated.

Thanks!

skeggse, 2018-03-23:
This sounds great!

FrancoisGrondin, 2018-03-24:
Ok, so I've done some changes to ODAS to allow using multiple spatial filters, which can be used to squeeze the search space to an arc shape to handle linear arrays. I've created a config file for the PS eye, though I don't have the actual hardware to test this configuration. Here is a small figure to illustrate what's being done:

[image]

So both the spatial filters pointing in the +/- z direction squeeze the search to a surface near the azimuth. The microphone directivity in the + x direction then ensures the source is only detected in front of the PSeye and not behind. I've pushed the new changes to the branch [spatialfilters](https://github.com/introlab/odas/tree/spatialfilters). Could you eventually test this new configuration and tell me if it does what you need? I'll then merge to master.

Cheers,

skeggse, 2018-03-26:
I'm super busy over the next few days. I'll try to get back to you on Friday.

FrancoisGrondin, 2018-04-04:
@skeggse Any success with your linear array? :)

skeggse, 2018-04-04:
Yikes, I'm sorry! I've been working on a write-up for my undergraduate capstone project, and haven't had time :'(

I'll try and block out some time tomorrow!

FrancoisGrondin, 2018-04-04:
No worries! Let me know if there is anything I can do to assist you!

Cheers

reynaldo, 2018-04-05:
Seems to work for me at 16kHz. Only tested to 'run' without failures with odascore though.

Commit here. Might issue a pull request after further testing:

https://github.com/reynaldo/odas/commit/f856ae742fb7eaa70d7b7228c696d91ff5ca1de7

If you are interested on using the eye as a reference, I might be able to ship one your way.

closed 2018-05-23 by FrancoisGrondin