introlab/odas #19: Spatial Filter configuration

taospartan, 2018-03-21:
Could you give some guidance on how to correctly use the spatial filter to block false detections?

Many thanks

Tao

FrancoisGrondin, 2018-03-22:
Do you know where the false detections come from? Reflections from walls or the floor?

taospartan, 2018-03-22:
It appears to be the floor

Many thanks

FrancoisGrondin, 2018-03-22:
Ok, then what you should do is aim the vector at the ceiling, and choose the angles to infer a gain of zero in the direction of the floor. Suppose the z-axis points to the ceiling:

```
    spatialfilter: {

        direction = ( +0.000, +0.000, +1.000 );
        angle = (80.0, 90.0);

};    
```
This configuration means that you allow a gain of 1 between angle 0 and 80, then there is a smooth transition in the gain from 1 to 0 and the angle range from 80 to 90, and finally the gain is 0 between 90 and 180 degrees. In other words, the system does not listen to what comes from the floor. Here's a drawing to illustrate what this parameter does:

[image]

taospartan, 2018-03-22:
That’s brilliant thank you very much, this is such a great piece of software!!

closed 2018-03-22 by FrancoisGrondin

alexengbloom, 2018-07-08:
Cool. Hello, @FrancoisGrondin . Does this also apply when using the matrix voice (microphones on the same plane) ?

link to matrix voice: https://www.matrix.one/products/voice