introlab/odas #21: Does ODAS work with a Delta 1010LT 8-channel audio card?

scotthasse, 2018-03-22:
Thanks for your very interesting project.  I know you are working on an installation tutorial, and I'll be very interested to see that.  In the mean time, I am trying to get ODAS to work with an M-Audio Delta 1010LT 8-channel input card.  Looking back at manyears, that hardware was seemingly supported, but I was wondering if that hardware will work with ODAS.  If so I'd be happy to work on sample configurations for this card.

Thanks again,

Scott

FrancoisGrondin, 2018-03-22:
Hi Scott,

Thank you for this input. Yes, if the card is working as a 8-channel input card, it should work just fine with ODAS. You'll have to specify the appropriate sample rate in the config, along with the number of bits per sample. For now you have to provide the card number in the config file, but this may change for each system according to the connected USB devices. I'm aware this is not the ideal way to do things, and for this reason I'm planning to change that so that you can provide the device name instead, and ODAS will find it automatically (see issue #1).

Please let me know if you experience any difficulties using the M-Audio Delta with ODAS.

Cheers,

Francois

scotthasse, 2018-03-26:
Thanks Francois.  I could use some assistance configuring the Delta 1010LT.  I was mistaken that the card is an 8-channel input.  It has 8 inputs, but they are devices rather than channels.  I am able to make a ~/.asoundrc file so that alsa sees the devices as individual channels, but it is not clear to me how I can configure odas to name the channels per microphone.  Also am I correct that the respeaker.cfg file is going to be the best starting point for the 1010lt?

Also, if aplay shows:

```
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: M1010LT [M Audio Delta 1010LT], device 0: ICE1712 multi [ICE1712 multi]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Intel [HDA Intel], device 0: STAC9200 Analog [STAC9200 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
would the corresponding configuration file be:

```
raw:
{

    fS = 16000;
    hopSize = 128;
    nBits = 16;
    nChannels = 8;

    # Input with raw signal from microphones
    interface: {
        type = "soundcard";
        card = 0;
        device = 0;
    }

}

```

Thanks,

Scott

scotthasse, 2018-03-26:
Sorry, arecord output is probably more relevant (but similar):
```
$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: M1010LT [M Audio Delta 1010LT], device 0: ICE1712 multi [ICE1712 multi]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Intel [HDA Intel], device 0: STAC9200 Analog [STAC9200 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

FrancoisGrondin, 2018-03-26:
Yes, assuming the card supports sample rate at 16000 samples/sec and 16 bits, this should be correct (you'd get an error message at runtime if the card does not support a specific configuration). Also, if you look at #24, it is now possible to call the sound card by its name instead of the number (since the number may change over time according to the usb devices connected to you computer).

scotthasse, 2018-03-27:
So I have made some progress configuring the Delta 1010LT.  FWIW it is a PCI card.  I am now getting the following when running odascore in verbose mode:

```
+--------------------------------------------+
|    ODAS (Open embeddeD Audition System)    |
+--------------------------------------------+
| Author:  Francois Grondin                  |
| Email:   francois.grondin2@usherbrooke.ca  |
| Website: introlab.3it.usherbrooke.ca       |
| Version: 1.0                               |
+--------------------------------------------+
| + Initializing configurations...... [Done] |
| + Initializing objects............. [Done] |
| + Launch threads................... [Done] |
| + Threads running.................. Source hops: Cannot set channel count: Invalid argument
```
I've attached the config file to this issue:
[delta1010lt.cfg.txt](https://github.com/introlab/odas/files/1852577/delta1010lt.cfg.txt)

What might be becoming relevant now is that the PCI device is a single device and the 8 mono input channels are not exposed as channels.  I can make individual devices for each mono input via a custom `~/.asoundrc` file:

```
pcm.one_channel {
    @args [ CHANNEL ]
    @args.CHANNEL { type integer }
    type dsnoop
    ipc_key 20130206
    slave {
        pcm "hw:0"
        channels 8
        rate 44100
    }
    bindings [ $CHANNEL ]
}
```

that can then each be successfully accessed via, for instance:

```
for channel in 0 1 2 3 4 5 6 7 ; do
echo "channel: ${channel}"
arecord --device=one_channel:${channel} --duration=2 -f S32_LE -c 1 --rate=44100 test${channel}.wav
done
```

but that requires me to specify the device name individually for each mono input channel.  I am not sure how to make that work with the odas configuration file.

I am not really sure if this is the correct approach at all for interfacing this card to odas, so any advice is greatly appreciated.

Thanks,

Scott

scotthasse, 2018-03-27:
I have also updated to the latest code to try the soundcard_name, referencing the pcm name:

```
raw:
{

    #fS = 16000;
    fS = 44100;
    hopSize = 128;
    #nBits = 16;
    nBits = 32;
    nChannels = 8;

    # Input with raw signal from microphones
    interface: {
        type = "soundcard_name";
        devicename = "one_channel";
    }

}

```

but I still get the following:

```
+--------------------------------------------+
|    ODAS (Open embeddeD Audition System)    |
+--------------------------------------------+
| Author:  Francois Grondin                  |
| Email:   francois.grondin2@usherbrooke.ca  |
| Website: introlab.3it.usherbrooke.ca       |
| Version: 1.0                               |
+--------------------------------------------+
| + Initializing configurations...... [Done] |
| + Initializing objects............. [Done] |
| + Launch threads................... [Done] |
| + Threads running.................. Source hops: Cannot set channel count: Invalid argument

```

FrancoisGrondin, 2018-03-27:
Hum that's a bummer. The warning message you get comes directly from ALSA not being able to open the device with the provided number of channels. You said this worked with ManyEars previously right? With ManyEars I was using RtAudio, with sort of wrap ALSA. Maybe RtAudio is doing something to allow using your sound card with Alsa. Out of curiosity, have you looked at the Alsamixer? Maybe there is a workaround with Jack audio, though I find this would be a sketchy patch...

scotthasse, 2018-03-27:
I have been able to make progress by configuring a multi channel virtual device in `~/.asoundrc` so that each of the 1010LT mono inputs shows up as a channel:

```
# .asoundrc for a Delta 1010LT to have individual record channels
#
# Create virtual devices
# JACK will need MMAP_COMPLEX support to use this. 
# ICE1712 chip has 12 capture channels and 10 playback channels.
# No. of channels in slaves must equal 12 for capture and 10 for playback
# otherwise "invalid argument" errors result.

pcm.multi_capture {
	type multi
	slaves.a.pcm hw:0 
	slaves.a.channels 12

# First 8 channels of first soundcard (capture)
 	bindings.0.slave a
 	bindings.0.channel 0
 	bindings.1.slave a
 	bindings.1.channel 1
 	bindings.2.slave a
 	bindings.2.channel 2
 	bindings.3.slave a
 	bindings.3.channel 3
 	bindings.4.slave a
 	bindings.4.channel 4
 	bindings.5.slave a
 	bindings.5.channel 5
 	bindings.6.slave a
 	bindings.6.channel 6
 	bindings.7.slave a
 	bindings.7.channel 7
    
}

ctl.multi_capture {
	type hw
	card 0
}
```

Then using the new `soundcard_name` feature in the odas config:

```
raw:
{

    #fS = 16000;
    fS = 44100;
    hopSize = 128;
    #nBits = 16;
    nBits = 32;
    nChannels = 8;

    # Input with raw signal from microphones
    interface: {
        type = "soundcard_name";
        devicename = "multi_capture";
    }

}
```

I can run `odascore` and see the following output (the program does not terminate):

```
+--------------------------------------------+
|    ODAS (Open embeddeD Audition System)    |
+--------------------------------------------+
| Author:  Francois Grondin                  |
| Email:   francois.grondin2@usherbrooke.ca  |
| Website: introlab.3it.usherbrooke.ca       |
| Version: 1.0                               |
+--------------------------------------------+
| + Initializing configurations...... [Done] |
| + Initializing objects............. [Done] |
| + Launch threads................... [Done] |
| + Threads running.................. 
```

I am not sure at this point how to tell if something productive is happening though.  I have attempting to integrate odas_web, but I see the following error after configuring the odas_web portions of the configuration file and using odas_web to spawn odascore:

```
| + Threads running.................. Sink hops: Cannot connect to server
```
is there perhaps some port mismatch, e.g. 10001 vs 10010?

Sorry if what I am doing does not make sense, I am new to most of the audio configuration.

Thanks,

Scott

FrancoisGrondin, 2018-03-27:
Hi scott,

Glad to see you are able to connect to the sound card :) If you are calling ODAS and the input comes from a sound card, it's okay for the program not to terminate, because the sound card is always feeding the system with new samples, and the library keeps processing the new samples. If you want to terminate the program, you can hit CTRL+C. The library will catch the signal, deallocate memory and free the sound card.

Now regarding the new error you have with ODAS web, it seems the library is not able to talk to ODASweb. I'm adding @GodCed to this thread, since he worked on the odasweb interface. Cedric, can you provide any feedback to Scott regarding this error he gets? Thanks!

scotthasse, 2018-03-27:
Thanks again Francois.  @GodCed I will attach my latest configuration.  I am happy to open an issue on odas_web if that would be more straightforward, just let me know.  I've followed the instructions at `https://github.com/introlab/odas_web`, although I am not sure what "The ODAS library must be built to use ODAS Studio. " means.  I just followed the stock instructions for building odascore.  When starting odas_web, I see the following output:

```
$ npm start

> ODAS_Desktop@0.2.0 start /home/shasse/odas/odas_web
> electron .

server listening to {"address":"::","family":"IPv6","port":9000}
server listening to {"address":"::","family":"IPv6","port":9001}
server listening to {"address":"::","family":"IPv6","port":10000}
server listening to {"address":"::","family":"IPv6","port":10010}
```

and I have made a small wrapper script around odascore so I can capture verbose output to a file:

```
$ cat odascore-verbose.sh 
/home/shasse/odas/odas/bin/odascore -v $* | tee /tmp/odas-verbose.log
```

I call that script as the "ODAS Core" from the GUI.  After configuring that script as the binary and a hopefully appropriate configuration file:

[delta1010lt-odas_web.cfg.txt](https://github.com/introlab/odas/files/1853190/delta1010lt-odas_web.cfg.txt)

I see the following from the verbose odascore logging:

```
+--------------------------------------------+
|    ODAS (Open embeddeD Audition System)    |
+--------------------------------------------+
| Author:  Francois Grondin                  |
| Email:   francois.grondin2@usherbrooke.ca  |
| Website: introlab.3it.usherbrooke.ca       |
| Version: 1.0                               |
+--------------------------------------------+
| + Initializing configurations...... [Done] |
| + Initializing objects............. [Done] |
| + Launch threads................... [Done] |
| + Threads running.................. Sink hops: Cannot connect to server
```

I see the ports listed do not quite match up and am wondering if that is the issue.

scotthasse, 2018-03-27:
In fact I changed port 10001 to port 10010 in the odas configuration and now I see actual data coming into the GUI for a bit, until one screen's worth of data passes.  I think that represents a documentation bug at `https://github.com/introlab/odas_web`:

```
postfiltered: {
  fS = <SAMPLE RATE>;
  hopSize = 512;
  nBits = 16;        

  interface: {
    type = "socket";
    ip = "<IP>";
    port = 10001;
  }        
};
```

After making that change I then see the following in the verbose output:

```
+--------------------------------------------+
|    ODAS (Open embeddeD Audition System)    |
+--------------------------------------------+
| Author:  Francois Grondin                  |
| Email:   francois.grondin2@usherbrooke.ca  |
| Website: introlab.3it.usherbrooke.ca       |
| Version: 1.0                               |
+--------------------------------------------+
| + Initializing configurations...... [Done] |
| + Initializing objects............. [Done] |
| + Launch threads................... [Done] |
| + Threads running.................. [Done] |
| + Free memory...................... [Done] |
+--------------------------------------------+
```

Does odas_web terminate odascore after some about of time?

Thanks!

Scott

FrancoisGrondin, 2018-03-27:
Just to make sure I understand: you launched odasweb, then odas, and then odas stopped after a while? Or it stopped when you closed odasweb?

GodCed, 2018-03-27:
Hi, @scotthasse you indeed spotted a typo in odas_web wiki. I'll correct it right now.
As for "the ODAS library must be built..." it simply means that odascore must be built (what you did) to use odas_web, as it doesn't embed odas directly.

Regarding unexpected closure of ODAS, it's not suppose to quit by itself, so something wrong is happening either in ODAS or in odas_web.

taospartan, 2018-03-27:
> I know sometimes ODAS may shutdown if there is a buffer overflow, meaning it cannot process in real-time all the frames coming from the sound card. ODAS is lightweight and should run easily, but I'm curious to know which hardware you are using to run it and odasweb?

Should the port be 10010?

I may be mistaken

GodCed, 2018-03-27:
Yes, the correct port is 10010 for the post-filtered audio, although you can change it the odas_web code if its more convenient for you.

scotthasse, 2018-03-27:
I am starting odas_web and then using that GUI to initiate odascore.  I encounter the same "odascore terminating" issue if I leave odas_web running and start odascore from a separate terminal using the configuration file that includes the odas_web config.  It usually runs about 16 seconds before terminating.

If I use a configuration file that does not have the odas_web configuration, most of the time odascore runs indefinitely (10 minutes at least) without terminating.  

The hardware is quite old, a Dell Optiplex 210L that I had sitting around: 

https://www.dell.com/downloads/global/products/optix/en/spec_optix_210l_en.pdf

I can try to dig up something newer, but it might take a bit.  In the mean time, how do I tell if things are functional without using the odas_web interface?

Thanks,

Scott

FrancoisGrondin, 2018-03-27:
Are you transmitting with sockets over a WiFi connection or an ethernet cable?

scotthasse, 2018-03-27:
I am using 127.0.0.1, so the local loopback interface.  The external interface is an ethernet cable.

FrancoisGrondin, 2018-03-27:
Let's try something: in demo/objects.c at line 765, could you change the following line

`objs->nMessages = 100;`

for

`objs->nMessages = 1000;`

and compile again, and then tell me if this makes a difference?

Thank you!

GodCed, 2018-03-27:
If that doesn't work I think you are experiencing a performance issue on your machine, as ODAS quits when it can't process in real time. ODAS by itself is lightweight, but odas_web is more ressource hungry. I seems to me that odas_web takes all the processing power, forcing ODAS to quit.

I would suggest you take a look at your machine usage monitor when it manages to run. You can also visualize ODAS output in your terminal, without odas_web. Simply edit your configuration file using
```
tracked: {
  format = "json";
  interface: {
    type = "terminal";
  };
};
```
as an interface. You can set the other interfaces to `type = "blackhole"`

scotthasse, 2018-03-27:
After changing `objs->nMessages = 1000;` it runs for between 3 minutes 30 seconds and 4 minutes 30 seconds now before terminating rather than ~16 seconds.  So it did seem to make a significant difference.  I'll try running the odas_web on different remote hardware, but unfortunately probably won't get to that for a couple of days with other responsibilities I have.

Thanks,

Scott

FrancoisGrondin, 2018-03-27:
Hi Scott,

Ok so that explains everything. Your computer is not able to run both odas and odasweb in real-time. Odas can run by itself, but not with odasweb running at the same time. When we increase the size of the buffer, odas still accumulates unprocessed frames from the sound card, but it takes more time before overflow because the buffer size is larger. Here are your options:

1) Display the results in the terminal instead of using odasweb
2) Run odas on your machine, and odasweb on another machine, and connect these with the socket
3) Change the full setup on a faster machine, and run both odas and odasweb

Cheers

closed 2018-03-27 by FrancoisGrondin

scotthasse, 2018-03-27:
Thanks so much, I should have results of doing one of these by the end of the week.

Scott

scotthasse, 2018-03-28:
Just to confirm that running the odas_web on separate, more capable hardware did indeed resolve the terminating issue.

GodCed, 2018-03-28:
I'm glad to hear that. Thanks for your feedback!