introlab/odas #25: Tell us what you do with ODAS

doumdi, 2018-03-23:
Please reply to this message and send us a link to your project if you use ODAS!

Regards,

The ODAS Team!

chengl2, 2018-03-26:
I want to use 'Sound Source Localization' to estimate elevation and azimut on my project.

I want to know how the ssl works.

my board has 8  microphones, looks like:

                 -
                 -
                 -
                 -
    -      -         -     -

Does this odas support this "T" microphone array?

thanks.

FrancoisGrondin, 2018-03-26:
Hi,

Yes it does. Please open a new issue so I could assist you with your configuration.

Best

chengl2, 2018-03-26:
OK, I've opend a new issue here:
https://github.com/introlab/odas/issues/27

scotthasse, 2018-03-27:
I am in an experimental phase of sound source localization to identify locations of specific sounds.  Depending on how well it works, there are several applications I am interested to experiment with.