introlab/odas #28: Potential Sorces Energy Range

taospartan, 2018-03-26:
Is it possible to restrict potential sources by their energy within odas? There is a slider with odas_web but can this be set in the odas cfg?

many thanks

FrancoisGrondin, 2018-03-26:
Sure you can do this. In the tracking module, you have the following parameters:

```
active = (
    { weight = 1.0; mu = 0.4; sigma2 = 0.0025 }
);

inactive = (
    { weight = 1.0; mu = 0.25; sigma2 = 0.0025 }
);
```

These stand for the distribution of energy for active and inactive sources. In this case, the active source energy follows a normal distribution with mean 0.4 and variance 0.0025. The inactive source has the same variance but a lower mean. If you want to make the system less subject to false detections, you can increase both mus. Note that there is a parameter weight: in this case it is left to 1 because you have a single gaussian. However, you could use a Gaussian Mixture Model (GMM) to represent the energy distribution of active and inactive sources, and have multiple entries, like this for instance:

```
active = (
    { weight = 0.8; mu = 0.4; sigma2 = 0.0025 },
    { weight = 0.2; mu = 0.6; sigma2 = 0.0015 },
);
```

In this case you must ensure the sum of weights are equal to 1. According to my experience, a single gaussian usually does the work, but I wanted to leave the option to use a GMM if ever needed.

taospartan, 2018-03-26:
Many thanks:)

closed 2018-03-26 by taospartan