introlab/odas #29: About "odascore binary"

bng025, 2018-03-26:
Apologies for asking something like this, but in this line "_For local execution, browse the **odascore binary** and the configuration file in the ODAS Control pane._", where can I locate the "odascore binary file?

FrancoisGrondin, 2018-03-26:
No need to apologize, there is no stupid question :) 

The odascore is the binary file that will be generated once you've compile the library. To do so, please refer to the installation page: https://github.com/introlab/odas/wiki/Installation

The odascore file is generated automatically after you've call the `make` command.

I hope that helps!

Cheers

closed 2018-03-27 by FrancoisGrondin

bng025, 2018-03-28:
Can the program run on Windows? If so, where can i download the external libraries and where do I place them?

FrancoisGrondin, 2018-03-28:
Not yet. I need to prepare a distribution for Windows and MacOS. I'll open a new issue. For now it can only run on linux.

Jimmy-Hu, 2018-08-21:
Hello,

After I built the library, I only saw the executable files "odaslive" and "odasserver". So, I wonder that the above-mentioned binary file "odascore" is referring to the file named "odaslive" ?

taospartan, 2018-08-21:
Odascore is from the Phoenix branch

Jimmy-Hu, 2018-08-21:
Thanks for your reply. I change to the Phoenix branch and rebuild the
project. However, there still only two executable files, "odaslive" and
"odasserver". Is there any step I lost?

> Odascore is from the Phoenix branch
>
> Sent from my iPad
>
> >
> > Hello,
> >
> > After I built the library, I only saw the executable files "odaslive"
> and "odasserver". So, I wonder that the above-mentioned binary file
> "odascore" is referring to the file named "odaslive" ?
> >
>
> <https://github.com/introlab/odas/issues/29#issuecomment-414821105>, or mute
> the thread

-- 
Chen-Hung Hu (胡鎮洪)
國立臺北科技大學 電子工程系
National Taipei University of Technology. Taipei, Taiwan
Department of Electronic Engineering
Mobile:+886-988-721737
Email:t103360021@ntut.org.tw
Address:
1, Sec. 3, Zhongxiao E. Rd., Taipei 10608 Taiwan, R.O.C.
10608 台北市忠孝東路三段一號

> Odascore is from the Phoenix branch
>
> Sent from my iPad
>
> >
> > Hello,
> >
> > After I built the library, I only saw the executable files "odaslive"
> and "odasserver". So, I wonder that the above-mentioned binary file
> "odascore" is referring to the file named "odaslive" ?
> >
>
> <https://github.com/introlab/odas/issues/29#issuecomment-414821105>, or mute
> the thread

-- 
Chen-Hung Hu (胡鎮洪)
國立臺北科技大學 電子工程系
National Taipei University of Technology. Taipei, Taiwan
Department of Electronic Engineering
Mobile:+886-988-721737
Email:t103360021@ntut.org.tw
Address:
1, Sec. 3, Zhongxiao E. Rd., Taipei 10608 Taiwan, R.O.C.
10608 台北市忠孝東路三段一號

taospartan, 2018-08-22:
Did you use git clone -b Phoenix <url>?

Jimmy-Hu, 2018-08-22:
Thanks for your reply. I re-clone the repository again with "git clone -b
Phoenix https://github.com/introlab/odas.git" and re-build the project.
There are five executable files "core", "filer", "formatter", "mapper", and
"sequencer".

-- 
Chen-Hung Hu (胡鎮洪)
國立臺北科技大學 電子工程系
National Taipei University of Technology. Taipei, Taiwan
Department of Electronic Engineering
Mobile:+886-988-721737
Email:t103360021@ntut.org.tw
Address:
1, Sec. 3, Zhongxiao E. Rd., Taipei 10608 Taiwan, R.O.C.
10608 台北市忠孝東路三段一號