introlab/odas #30: acoustic camera

taospartan, 2018-03-26:
Has odas ever been used as part of an acoustic camera?

FrancoisGrondin, 2018-03-26:
We did use odas in the past to add a tag on the image to display where the active sources were. I think what you want is different. If I understand your question, you would like to plot some sort of "heat layer" that contains the acoustic energy for all direction of arrivals, and to project it over the image. Is this what you would like to achieve?

taospartan, 2018-03-26:
Yes that is the idea, a heat map that allows the user to see the origin of the sound

FrancoisGrondin, 2018-03-26:
In this case, no, unfortunately, this feature is not implemented yet. If you feel this can be useful, this is something I'll implement.

taospartan, 2018-03-26:
I think this feature would greatly help many projects, as acoustic cameras can be outside the budget of most.

FrancoisGrondin, 2018-03-26:
Sure, I'll try to add this feature during my spare time in the coming days then.

taospartan, 2018-03-26:
You are very kind!😀

ketronium, 2020-06-12:
@taospartan did you have any luck building your acoustic camera?