introlab/odas #36: Error while using matrix_voice.cfg

luc2yj, 2018-04-03:
I have followed your wiki page and encountered the following error.
How do I solve it?

**command below:**
pi@raspberrypi:~/git/odas/bin $ ./odascore -c ../config/matrix_voice.cfg
 
**gives following error:**
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM 8channel_voice
Source hops: Cannot open audio device 8channel_voice: No such file or directory

FrancoisGrondin, 2018-04-03:
Hi,

It seems that the sound card does is not recognized with the name "8channel_voice". I know @Slothologist has been using this card (and he also kindly provided us with the config file and the feature to allow calling sound card by names instead of numbers). @Slothologist, have you ever encountered this problem before?

Thanks,

Slothologist, 2018-04-04:
Hi,
The PCM given in the config file was a manual addition to my alsa config file (/etc/asound.conf or ~/.asoundrc), so this may very possibly not be a bug within the odas software.
Using Matrix Voice with alsa is a bit tricky, as different setups of the MatrixIo software (and sometimes even different versions of it...) result in different alsa configs/ ways to interact with the microphones.
Depending on your matrix software version and/or alsa config you may not need such a virtual device. For me it was necessary, as the Voice was not one alsa device with 8 channels but rather a device with 8 subdevices  (each having 1 channel). 

@luc2yj if you could provide your alsa config file(s), i could provide you with the corresponding addition to it/ a working soundcard_name

luc2yj, 2018-04-05:
Thanks for helping. Here are the following files.
--------------------
/etc/asound.conf
--------------------
pcm.sc {
    type hw
    card 1
}

pcm.mic_channel0 {
  type file
  file "/tmp/matrix_micarray_channel_0"
  infile "/tmp/matrix_micarray_channel_0"
  format "raw"
  slave {
    pcm sc
  }
}

pcm.mic_channel1 {
  type file
  file "/tmp/matrix_micarray_channel_1"
  infile "/tmp/matrix_micarray_channel_1"
  format "raw"
  slave {
    pcm sc
  }
}

pcm.mic_channel2 {
  type file
  file "/tmp/matrix_micarray_channel_2"
  infile "/tmp/matrix_micarray_channel_2"
  format "raw"
  slave {
    pcm sc
  }
}

pcm.mic_channel3 {
  type file
  file "/tmp/matrix_micarray_channel_3"
  infile "/tmp/matrix_micarray_channel_3"
  format "raw"
  slave {
    pcm sc
  }
}

pcm.mic_channel4 {
  type file
  file "/tmp/matrix_micarray_channel_4"
  infile "/tmp/matrix_micarray_channel_4"
  format "raw"
  slave {
    pcm sc
  }
}

pcm.mic_channel5 {
  type file
  file "/tmp/matrix_micarray_channel_5"
  infile "/tmp/matrix_micarray_channel_5"
  format "raw"
  slave {
    pcm sc
  }
}

pcm.mic_channel6 {
  type file
  file "/tmp/matrix_micarray_channel_6"
  infile "/tmp/matrix_micarray_channel_6"
  format "raw"
  slave {
    pcm sc
  }
}

pcm.mic_channel7 {
  type file
  file "/tmp/matrix_micarray_channel_7"
  infile "/tmp/matrix_micarray_channel_7"
  format "raw"
  slave {
    pcm sc
  }
}

pcm.mic_channel8 {
  type file
  file "/tmp/matrix_micarray_channel_8"
  infile "/tmp/matrix_micarray_channel_8"
  format "raw"
  slave {
    pcm sc
  }
}

pcm.!default
{
  type asym
  playback.pcm {
    type hw
    card 0
    device 0
  }
  capture.pcm {
    type file
    file "/tmp/matrix_micarray_channel_0"
    infile "/tmp/matrix_micarray_channel_0"
    format "raw"
    slave {
        pcm sc
    }
  }
}

-------------------
~/.asoundrc
-------------------
pcm.!default {
	type hw
	card 0
}

ctl.!default {
	type hw
	card 0
}

luc2yj, 2018-04-05:
Setting the devicename to "sc" drops an error as below.
pi@raspberrypi:~/git/odas/bin$ ./odascore -c ../config/matrix_voice.cfg
Source hops: Cannot set channel count: Invalid argument

Slothologist, 2018-04-06:
Hi,
your /etc/asound.conf looks just like one of my old confs. You also have the 8 seperate subdevices (pcm.mic_channel0 to 8), and lack a virtual device to bundle them. 
But those are 9 mic_channel devices you might ask? Yes. And i have no idea why the is a pcm.mic_channel8 or what its purpose is. I first consciously encountered it examining your asound.conf, though it appears in old configs of mine as well. I also could not reproduce it, as a fresh setup of a pi with my matrix voice did not result in such a /etc/asound.conf. I am deeply confused. 

However, just adding a virtual device to your /etc/asound.conf could solve your problem. Try appending the following lines to your /etc/asound.conf:

> pcm.8channel_voice {
  type multi
>
>  slaves.a.pcm "mic_channel0"
  slaves.a.channels 1
  slaves.b.pcm "mic_channel1"
  slaves.b.channels 1
  slaves.c.pcm "mic_channel2"
  slaves.c.channels 1
  slaves.d.pcm "mic_channel3"
  slaves.d.channels 1
  slaves.e.pcm "mic_channel4"
  slaves.e.channels 1
  slaves.f.pcm "mic_channel5"
  slaves.f.channels 1
  slaves.g.pcm "mic_channel6"
  slaves.g.channels 1
  slaves.h.pcm "mic_channel7"
  slaves.h.channels 1
>
>  bindings.0.slave a
  bindings.0.channel 0
  bindings.1.slave b
  bindings.1.channel 0
  bindings.2.slave c
  bindings.2.channel 0
  bindings.3.slave d
  bindings.3.channel 0
  bindings.4.slave e
  bindings.4.channel 0
  bindings.5.slave f
  bindings.5.channel 0
  bindings.6.slave g
  bindings.6.channel 0
  bindings.7.slave h
  bindings.7.channel 0
}

This should create a new pcm device which taps into the existing alsa devices ("slaves.a.pcm "mic_channel0" " etc ) and bundles them to a single, 8-channel device ("bindings.0.slave a" etc).
Then try this new device for odas. I hope it helps!

luc2yj, 2018-04-11:
@Slothologist , Thank you for your help. The additional code removed all the errors. However, with the current configuration I am not able record any sound; thus I am not able to localize any sound source. Are you also using Matrix Voice + Raspberry Pi Model B? If so, does sound localization using ODAS work well on your end?

Slothologist, 2018-04-12:
I was unfortunately not able to make even basic recording via alsa work on the Voice. The Matrix Voice demo works for me too, but that does not use alsa afaik.

My use case was to use Matrix Voice with a RasPi 3B as a supplement for the pepper robot in a realtime scenario, as bringing any somewhat complex software on the pepper is somewhat of a hassle and comes with a big overhead if you just want to explore a new lib. However, getting the Voice to work properly was at least an equal hassle. In the end, due to time constraints, i moved on from the Voice.

If you are determined to making this work, i suppose you should first check whether or not you are able to record via alsa at all (for example via "arecord -f S16_LE -D mic_channel0 first_channel.wav") and then if the recorded audio makes sense (for example via audacity: If imported audio does not look right, eg has no real waveform, the recoding may be broken) If not, work from there.

Sorry that i can not be of more help.

yoelrc88, 2018-04-12:
@luc2yj and @Slothologist, I manage to make odas working with the Matrix voice board. There is now a kernel package available that support audio. You would have to install the Matrix basic software  ([matrix-creator-init](https://github.com/matrix-io/matrix-creator-init)) and then the kernel modules ([matrixio-kernel-modules](https://github.com/matrix-io/matrixio-kernel-modules)).

You can follow:
```
# Add repo and key
curl https://apt.matrix.one/doc/apt-key.gpg | sudo apt-key add -
echo "deb https://apt.matrix.one/raspbian $(lsb_release -sc) main" | sudo tee /etc/apt/sources.list.d/matrixlabs.list

# Update packages and install
sudo apt-get update
sudo apt-get upgrade

# Reboot
sudo reboot

# Connect again and install init
sudo apt install matrixio-creator-init

# Installation MATRIX Packages
sudo apt install matrixio-kernel-modules

# Reboot
sudo reboot
```
Thats all you need from the MATRIX side to make the board and audio work.
You can confirm that audio is working by recording and playing, use:
```
arecord -l
arecord -D hw:2,0 -r 16000  -f S16_LE test.wav && aplay test.wav
```
Then you need to update the config file for the voice:
```
...
    # Input with raw signal from microphones
    interface: {
        type = "soundcard";
        card = 2;
        device = 0;
    }
...
```
This would use a diferent type of interface and select the correct card and device.

This should be all you need. Let me know how it goes.

BTW @FrancoisGrondin, this project ODAS is amazing. Everything just worked! I am digging into it more to see how can I make a simpler DoA in terms of processing requirements (RPi 3 cpu was at 100%) so I can use it for a demo with the Matrix Voice. Looking forward to more documentation. Are the algorithms explained somewhere? Thank you.

FrancoisGrondin, 2018-04-12:
Thank you @yoelrc88 for your feedback! I'm surprised you are at 100% cpu however :O I did run everything on the RPi3 with 16 microphones and I still had CPUs available. I should investigate that. Can you provide me with more details? I have two papers under review which explains the math behind the system, but I'm waiting for reviewers' feedback before putting this online. Moreoever, I'm also writing a paper which describes the library itself, and I'll put it on Arxiv as soon as the first version is ready. I'll then add some additional info in a tutorial section (sorry for this delay... I'm working full time as a postdoc on other projects but I want to get documentation asap because many people are asking for it).

yoelrc88, 2018-04-13:
Maybe the CPU usage in my test was related to the setup of the Filter/Tracking algorithm in the config file for MATRIX Voice. Which config file were you using for the 16 mics you mentioned? Maybe I can compare both setups and find some differences. I will test again and if still using that much CPU I will create a new issue so we can follow up there. Thank you!

closed 2018-04-26 by FrancoisGrondin