introlab/odas #37: Add AEC

FrancoisGrondin, 2018-04-05:
Need to add auto echo-cancellation for video-conference application (or voice commands when the device is playing music).

xiongyihui, 2018-04-09:
@FrancoisGrondin I'm working an AEC project using SpeexDSP AEC algorithm. It can be used with ODAS as a file source. The structure is like:

```
+-------------------+  +-------------------+
|  PulseAudio       |  |  PulseAudio       |
|  module-pipe-source  |  module-pipe-sink |
+----------+--------+  +-------+-----------+
           |                   |             +------------+
           |                   |             |ALSA        |
           v                   v             |file plugin |
    +------+-------+    +------+-------+     +----+-------+
    |    FIFO      |    |    FIFO      |          |
    |/tmp/ec.output|    | /tmp/ec.input| <--------+
    +------+-------+    +------+-------+
           ^                   |
           |                   |
           |                   |
           |                   |
        +--+--+                |
        | AEC | <--------------+
        +--+--+                |
           ^                   |
           |                   v
    +------+-------+    +------+-------+
    |  hw:x,y      |    |  hw:x,y      |
    |  plughw:x,y  |    |  plughw:x,y  |
    +--------------+    +--------------+
```

FrancoisGrondin, 2018-04-10:
@xiongyihui Thank you for the input! Will definitely look at it :)

jan4984, 2018-05-29:
any plan please?

FrancoisGrondin, 2018-05-29:
This is something in the pipeline, but we are currently implementing other modules too! Doing our best to get something asap!

zuowanbushiwo, 2018-07-24:
@FrancoisGrondin  This project is great，with this We can do smart speakers ourselves. I am looking forward to your AEC。 thanks！

taospartan, 2018-08-13:
I think this would be a great feature also!

jamesshao8, 2018-08-14:
I am also looking forward to this echo cancellation