introlab/odas #39: using odas with the respeaker 4 mic array

mogar, 2018-04-09:
I'm trying to get odas running on a raspberry pi with my respeaker 4 mic array. I think the mic array is set up correctly, since I can record audio from all microphones using arecord.

I have odas_web studio 0.1 running locally, and I'm trying to run odascore with the default respeaker config file. I verified that the card number and device number match what I get from "arecord -l".

When I run it, I get the error:

`Source hops: Cannot set sample format: Invalid argument`

The odas_web studio does show that there are two client connections being formed (on ports 50498 and 41114), but they close immediately.

What do I have to do to get this running?

GodCed, 2018-04-09:
Try replacing the format from 16 to 32 bits in the raw input config.
Line 12 is ``nBits = 16;``
Replace with ``nBits = 32;``

mogar, 2018-04-09:
Looks like that worked. Thanks.

closed 2018-04-09 by mogar

ELELUABDULSALAM, 2022-01-31:
this is not working for, got the same error

commit 2f64a8b6 (2018-04-09): Fix #39 - wrong bit format from S16_LE to S32_LE