introlab/odas #42: some separated recorded wav file is empty

papayazm, 2018-04-20:
The problem happens randomly.  Some recorded file is OK ,but some of them are empty. Attached is my empty file.

FrancoisGrondin, 2018-04-20:
I'm not sure I understand: these files come from the odas_web interface?

GodCed, 2018-04-20:
Could you give more details about your test setup and scenario in order for us to replicate and investigate the issue? Are the empty files always extremely short, say less then 2 seconds, like the one you submitted?

papayazm, 2018-04-23:
Yes, the empty file is from the odas_web recorded file directory. I am using MiniDSP 8 channel.  I say the number from 1 to 20, maybe like "1,2,3,4",keep silence for 5 seconds, "5,6",  keep silence for 5 seconds, "7,8,9" .... "17,18,19,20". When I keep silence for 5 seconds, a new recorded file will be created. So I have some recorded files after I count number from 1 to 20. But some of them is empty, maybe "5,6", maybe "7,8,9", happened randomly. And some of them is OK.

FrancoisGrondin, 2018-04-26:
@papayazm 

This seems to be due to the fact that the system is not able to track all digits over time. Suppose 1,2,3,4 are detected, then there is a silence, so there is not tracking. Now if 5,6 does not trigger a new source tracking, you won't get any separated stream. Do you also output the tracking info to see if this is what is happening?

Francois

papayazm, 2018-04-26:
@FrancoisGrondin 
Thanks. The problem is easily come out when in a noise environment.  So It is OK when in clear environment.

closed 2018-04-26 by papayazm