introlab/odas #45: Changing the value of samplerate.mu.

luc2yj, 2018-04-25:
I see that it is recommended to leave samplerate.mu as 16000 as shown below.

samplerate.mu | uint | You should leave this parameter at 16000

I have assigned  48000 to fS of raw section and am planning to increase it to 96000. Am I still recommended to leave samplerate.mu to 16000? What is the effect of 16000 and why is it recommended to keep it at 16000?

Thanks!

FrancoisGrondin, 2018-04-26:
Hi!

If you change samplerate.mu to a larger value, you'll have to also increase the frameSize to capture the same duration of signal in msec. If you do so, the processing power required will increase. The TDOA discretization will also change, though increasing the sample rate is usually a good thing regarding discretization. Can you tell me why you'd like to increase the sample rate? Is it because you want to capture high frequency components? Have you also considered your microphone bandwidth?

luc2yj, 2018-04-26:
the microphones that I am using are close together 2.5cm ~ 8cm. 
Sound travels 2.5cm in 0.0000735 seconds while sample period is 0.0000625 (16000hz) seconds. To increase accuracy, I thought it would be better to increase the sampling rate. The microphone can support up to 96000hz.

FrancoisGrondin, 2018-04-26:
@luc2yj 

I have a solution for you: in the ssl section of the config file, you have this parameter:

`interpRate = 1;`

It is the interpolation rate. If you set it to 2, it's like sampling at 2x16000 samples/sec, to 3 is like 3x16000 samples/sec, etc.

This way you can increase accuracy without increasing the input sample rate that will propagate in all the pipeline some very large frames padded with zeros.

luc2yj, 2018-04-26:
Great. I will try this solution!

closed 2018-04-26 by FrancoisGrondin

luc2yj, 2018-05-23:
I understand that I can get better accuracy by increasing the interpolation rate. Should I change the samplerate.mu if I want to capture high frequency sounds?