introlab/odas #47: Discretization

taospartan, 2018-04-26:
Which settings can be adjusted to increase Discretization?

many thanks

taospartan, 2018-04-26:
> @taospartan Just to make sure, you are referring to the Search space discretization?

Yes:)

FrancoisGrondin, 2018-04-26:
Ok, you can do this here in the SSL module:

`    scans = (
        { level = 2; delta = -1; },
        { level = 4; delta = -1; }
    );`

The discretization level is defined by "level". The number of points on the sphere are given by (20 x 4^L)/2 + 2, where L is the level:

level = 0 -> 12 points
level = 1 -> 42 points
level = 2 -> 162 points
level = 3 -> 642 points
level = 4 -> 2562 points
level = 5 -> 10242 points
...

You see two levels in this example, level = 2 and = 4. This means the system first perform a coarse scan of space with 162 points, and then refine online the required points on the grid with 2562 points (this speeds up process as only a subset of the 2562 points grid is sampled). If you want to increase the discretization, you may add a third grid, with say level = 5. This should work. Be aware that you may also play with the interpRate parameter if you microphones are closed to each other, to reduce the discretization introduced while estimating TDOA.

Does this help?

Cheers

taospartan, 2018-04-26:
It sure does thank you !