introlab/odas #48: Filtering based on audio frequency.

luc2yj, 2018-04-30:
Hello, 

I am planning to add a filter in order to analyze audio signals within a certain range; e.g., over XXXhz and less than XXXhz. Where would be the best place to modify to add such feature?

Thanks,

Peter

FrancoisGrondin, 2018-05-03:
Hi Peter,

The spectral information for each channel is stored in the messages amsg_spectra/msg_spectra. If you want to modify the frequency range, maybe the neat solution would be to create a new module that accepts a spectra message, changes the frequencies, and output the modified spectra.

Is this for localization purpose?

Francois

chrisvwn, 2023-11-01:
I am interested in this as well. I am looking to be able to supply a frequency or range of frequencies and be able to locate the source.