introlab/odas #49: Where can I get the raw audio files specfied in the configs?

tigaworx, 2018-05-02:
Sorry for asking such a basic question, but I wondered if you guys shared the raw audio files that are specified in some of the config files in the odas/config directory somewhere. For example in xmos.cfg the file "meeting_recorded_raw_16.raw" is specified.
I just compiled the odas project and want to do a test run with some existing config file (and raw file) just to see what the program does when executed.

FrancoisGrondin, 2018-05-03:
Hmm I'd have to check where are these files. I could probably also create a directory with some dataset so you can run files recorded from different microphone arrays (and people would also be able to contribute with their own files). What do you think?

Cheers,

tigaworx, 2018-05-04:
Thank you FrancoisGrondin, for your answer. By now I managed to run odas with odas_web and the 8soundsUSB successfully. Thanks to the configuration page https://github.com/introlab/odas/wiki/configuration I managed to create my own config file for recording from the mics. 
In my opinion providing an example raw-file with some example config file would speed up the getting started part for fresh explores of the project though. So, a demo-directory with some dataset would be a nice-to-have.
Odas is an amazing project by the way!

closed 2018-05-04 by tigaworx

tigaworx, 2018-08-01:
@zuowanbushiwo You can find my config file in this thread [https://github.com/introlab/odas_web/issues/9]
I did not record any raw data though. I use the record interface of odas web.