introlab/odas #52: Using MATRIX Voice ... works amazing ... CPU usage over 100% :(

yoelrc88, 2018-05-11:
My setup is a MATRIX Voice board and a Raspberry Pi 3 model B. I am using the config file `/master/config/matrix_voice.cfg` as it is.

In my tests using threads, I am always getting CPU usage > 100%. I read in other issues that this is not normal. 

```bash
pi@odas-test:~/odas/bin $ ./odascore -vc ../config/matrix_voice.cfg
+--------------------------------------------+
|    ODAS (Open embeddeD Audition System)    |
+--------------------------------------------+
| Author:  Francois Grondin                  |
| Email:   francois.grondin2@usherbrooke.ca  |
| Website: introlab.3it.usherbrooke.ca       |
| Version: 1.0                               |
+--------------------------------------------+
| + Initializing configurations...... [Done] |
| + Initializing objects............. [Done] |
| + Launch threads................... [Done] |
| + Threads running.................. Sink hops: Could not send message.
pi@odas-test:~/odas/bin $ date
Fri May 11 00:04:45 UTC 2018
pi@odas-test:~/odas/bin $ ./odascore -vsc ../config/matrix_voice.cfg
+--------------------------------------------+
|    ODAS (Open embeddeD Audition System)    |
+--------------------------------------------+
| Author:  Francois Grondin                  |
| Email:   francois.grondin2@usherbrooke.ca  |
| Website: introlab.3it.usherbrooke.ca       |
| Version: 1.0                               |
+--------------------------------------------+
| + Initializing configurations...... [Done] |
| + Initializing objects............. [Done] |
| + Processing....................... [Done] |
| + Free memory...................... [Done] |
+--------------------------------------------+
|              Profiler summary              |
+--------------------------------------------+
| + Raw                                      |
|    - Source.......... 000.298 (003.30%)    |
|    - Connector....... 000.011 (000.12%)    |
| + Mapping                                  |
|    - Module.......... 000.010 (000.11%)    |
|    - Connector....... 000.007 (000.08%)    |
| + Resample                                 |
|    - Module.......... 000.786 (008.70%)    |
|    - Connector....... 000.009 (000.10%)    |
| + STFT                                     |
|    - Module.......... 000.110 (001.21%)    |
|    - Connector....... 000.020 (000.22%)    |
| + Noise                                    |
|    - Module.......... 000.145 (001.61%)    |
|    - Connector....... 000.008 (000.09%)    |
| + SSL                                      |
|    - Module.......... 003.328 (036.84%)    |
|    - Connector....... 000.005 (000.05%)    |
|    - Sink............ 000.189 (002.09%)    |
| + Target                                   |
|    - Injector........ 000.009 (000.10%)    |
|    - Connector....... 000.004 (000.05%)    |
| + SST                                      |
|    - Module.......... 000.090 (001.00%)    |
|    - Connector....... 000.008 (000.09%)    |
|    - Sink............ 000.084 (000.93%)    |
| + SSS                                      |
|    - Module.......... 000.470 (005.20%)    |
|    - Connector....... 000.011 (000.00%)    |
| + ISTFT                                    |
|    - Module.......... 000.182 (002.02%)    |
|    - Connector....... 000.010 (000.12%)    |
| + Resample                                 |
|    - Module.......... 000.077 (000.85%)    |
|    - Connector....... 000.008 (000.09%)    |
| + Volume                                   |
|    - Module.......... 000.015 (000.17%)    |
|    - Connector....... 000.007 (000.08%)    |
|    - Sink............ 000.150 (001.66%)    |
| + Classify                                 |
|    - Module.......... 002.206 (024.42%)    |
|    - Connector....... 000.004 (000.05%)    |
|    - Sink............ 000.004 (000.04%)    |
+--------------------------------------------+
| + TOTAL.............. 008.257 (091.39%)    |
|    - Module.......... 007.418 (082.11%)    |
|    - Connector....... 000.113 (001.25%)    |
|    - Source.......... 000.298 (003.30%)    |
|    - Sink............ 000.427 (004.73%)    |
+--------------------------------------------+
pi@odas-test:~/odas/bin $ date
Fri May 11 00:05:01 UTC 2018
pi@odas-test:~/odas/bin $ 

```
Pasted the outputs of both option, using threads and not using it. When using the single process version the odascore stops in less than 20 sec.

 @francoisferland, Any idea what could be happening here?

Is there any other output that I can share that gives you more info?
Thanks.
-Yoel

PD: The processing/tracking is amazing ... congrats on that!

FrancoisGrondin, 2018-05-11:
Thank you for the feedback! A few things:

1) Using a single thread, you're under 100%. This means multithread should work under 100% easily, unless something is eating up all processing power, which seems to be the case.
2) Seems that the sinks that receive the separated/postfiltered streams cause the problem. Can you change these to the following:

```
        interface: {
            type = "blackholde";
        }  
```

And then tell me if this solves your issue?

Cheers!

yoelrc88, 2018-05-11:
Thanks for the fast response @FrancoisGrondin, 

I found the separated label just in the `sss` block. I used the `blackhole` type instead but had the sample problem again:
- **odascore** exiting after few seconds when running it in a single thread
- **odascore** running well with multithreads but above 100% in CPU.

Please see below:

Change in the config file:
```
sss:
{
...
    separated: {

        fS = 16000;
        hopSize = 128;
        nBits = 16;        

        interface: {
            type = "blackhole";
            # type = "socket";
            # ip = "192.168.1.14";
            # port = 10000;
        }        

    };
...
};
```

Running **odascore** with `-s`:
```
pi@odas-test:~/odas/bin $ date
Fri May 11 16:42:24 UTC 2018
pi@odas-test:~/odas/bin $ ./odascore -vsc ../config/matrix_voice.cfg
+--------------------------------------------+
|    ODAS (Open embeddeD Audition System)    |
+--------------------------------------------+
| Author:  Francois Grondin                  |
| Email:   francois.grondin2@usherbrooke.ca  |
| Website: introlab.3it.usherbrooke.ca       |
| Version: 1.0                               |
+--------------------------------------------+
| + Initializing configurations...... [Done] |
| + Initializing objects............. [Done] |
| + Processing....................... [Done] |
| + Free memory...................... [Done] |
+--------------------------------------------+
|              Profiler summary              |
+--------------------------------------------+
| + Raw                                      |
|    - Source.......... 000.202 (003.32%)    |
|    - Connector....... 000.007 (000.12%)    |
| + Mapping                                  |
|    - Module.......... 000.006 (000.10%)    |
|    - Connector....... 000.005 (000.09%)    |
| + Resample                                 |
|    - Module.......... 000.529 (008.68%)    |
|    - Connector....... 000.006 (000.10%)    |
| + STFT                                     |
|    - Module.......... 000.071 (001.16%)    |
|    - Connector....... 000.014 (000.22%)    |
| + Noise                                    |
|    - Module.......... 000.095 (001.56%)    |
|    - Connector....... 000.005 (000.08%)    |
| + SSL                                      |
|    - Module.......... 002.257 (037.05%)    |
|    - Connector....... 000.004 (000.06%)    |
|    - Sink............ 000.129 (002.12%)    |
| + Target                                   |
|    - Injector........ 000.003 (000.06%)    |
|    - Connector....... 000.003 (000.05%)    |
| + SST                                      |
|    - Module.......... 000.038 (000.63%)    |
|    - Connector....... 000.006 (000.11%)    |
|    - Sink............ 000.054 (000.89%)    |
| + SSS                                      |
|    - Module.......... 000.189 (003.10%)    |
|    - Connector....... 000.007 (000.00%)    |
| + ISTFT                                    |
|    - Module.......... 000.118 (001.94%)    |
|    - Connector....... 000.007 (000.12%)    |
| + Resample                                 |
|    - Module.......... 000.051 (000.83%)    |
|    - Connector....... 000.005 (000.07%)    |
| + Volume                                   |
|    - Module.......... 000.010 (000.16%)    |
|    - Connector....... 000.005 (000.08%)    |
|    - Sink............ 000.061 (001.00%)    |
| + Classify                                 |
|    - Module.......... 001.511 (024.80%)    |
|    - Connector....... 000.003 (000.05%)    |
|    - Sink............ 000.003 (000.04%)    |
+--------------------------------------------+
| + TOTAL.............. 005.400 (088.66%)    |
|    - Module.......... 004.874 (080.03%)    |
|    - Connector....... 000.077 (001.26%)    |
|    - Source.......... 000.202 (003.32%)    |
|    - Sink............ 000.247 (004.05%)    |
+--------------------------------------------+
pi@odas-test:~/odas/bin $ date
Fri May 11 16:42:38 UTC 2018
```

Running it without `-s`
```
pi@odas-test:~/odas/bin $ ./odascore -vc ../config/matrix_voice.cfg
+--------------------------------------------+
|    ODAS (Open embeddeD Audition System)    |
+--------------------------------------------+
| Author:  Francois Grondin                  |
| Email:   francois.grondin2@usherbrooke.ca  |
| Website: introlab.3it.usherbrooke.ca       |
| Version: 1.0                               |
+--------------------------------------------+
| + Initializing configurations...... [Done] |
| + Initializing objects............. [Done] |
| + Launch threads................... [Done] |
| + Threads running.................. ^C[Done] |
| + Free memory...................... [Done] |
+--------------------------------------------+
pi@odas-test:~/odas/bin $ 

```

On other topic. Looking forward to understand better about the library. If you can share any docs, even if they are not complete would be awesome. I would like to run it maybe removing some of the parts or tweaking some of them ( ex tracking ) to be less CPU intensive. 

Final question: From your experience , what's the expected CPU % when using single and multi thread options?

Again, thanks a lot.

FrancoisGrondin, 2018-05-11:
I have calculated the respective % usage for SSL and SST modules:

1) SSL: About 15% of one core with 8 mics (but it also depends on the mic array shape, frame size, etc.)
2) SST: Less than 5% when using tracking based on Kalman filters and when tracking 4 sources simultaneously

For SSS, it depends on the method, but around 30% when separating 4 sources simultaneously.

In your case, the 100% CPU, maybe that's the sum of all threads? In other words, you have up to 400% CPU with 4 cores.

The doc is being written right now: I'm planning to submit a paper in 4-6 weeks, and I'll post the initial version on Arxiv, so you guys can read it already. I'm also modifying the code to disable modules not used (ex: classify in this case). That should be online soon, and thus you'll automatically free processing power!

Cheers

yoelrc88, 2018-05-11:
Ok, thanks for the answer. Now I can see that maybe there is no problem at all. I'll just use for now the multithread option. 

Luck with the paper! Looking forward to reading it.

closed 2018-05-11 by FrancoisGrondin