introlab/odas #57: Increase the silence

taospartan, 2018-05-18:
How can you increase the period of silence before a source stops being tracked?

Cheers

FrancoisGrondin, 2018-05-18:
Hey there!

Do increase the silence, you should play with the following parameters:

`N_inactive = ( 150, 200, 250, 250 );`

This tells you how many frames the period of silence should last before the source stops being tracked. In this case, there are four values, because we can track up to four sources. This means that if only one source is tracked, the silence period will be 150 frames. If two sources are tracked, it will be 200 frames. We usually increase the number of frames when they are more sources, because less observations are assigned to each source when many are competing.

taospartan, 2018-05-18:
That’s great, many thanks. I have a config for a 16 mic square array from minidsp if you are interested?

Tao

FrancoisGrondin, 2018-05-18:
Oh sure! Minidsp provided me with this array a few weeks ago, and I plan to write the code to create an acoustic image! Feel free to share the config if you want! Many thanks!

taospartan, 2018-05-18:
Wow sounds cool!

taospartan, 2018-05-20:
here is the mic layout, hope its right:)

mics = (
        
        # Microphone 1
        { 
            mu = ( -0.021, -0.063, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 90.0 );
        },

        # Microphone 2
        { 
            mu = ( -0.063, -0.063, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 90.0 );
        },

        # Microphone 3
        { 
            mu = ( -0.021, -0.021, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 90.0 );
        },

        # Microphone 4
        { 
            mu = ( -0.063, -0.021, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
           angle = ( 80.0, 90.0 );
        },

        # Microphone 5
        { 
            mu = ( -0.021, +0.021, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 90.0 );      
        },

        # Microphone 6
        { 
            mu = ( -0.063, +0.021, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
           angle = ( 80.0, 90.0 );       
        },

        # Microphone 7
        { 
            mu = ( -0.021, +0.063, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 90.0 );
        },
        
        # Microphone 8
        { 
            mu = ( -0.063, +0.063, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 90.0 );
        },

        # Microphone 9
        { 
            mu = ( +0.063, +0.063, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 90.0 );
        },

        # Microphone 10
        { 
            mu = ( +0.021, +0.063, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 90.0 );
        },

        # Microphone 11
        { 
            mu = ( +0.063, +0.021, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 90.0 );
        },

        # Microphone 12
        { 
            mu = ( +0.021, +0.021, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 90.0 );
        },

        # Microphone 13
        { 
            mu = ( +0.063, -0.021, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 90.0 );
        },

        # Microphone 14
        { 
            mu = ( +0.021, -0.021, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 90.0 );
        },

        # Microphone 15
        { 
            mu = ( +0.063, -0.063, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 90.0 );
        },

        # Microphone 16
        { 
            mu = ( +0.021, -0.063, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 90.0 );
        }

    );

FrancoisGrondin, 2018-05-21:
Thank you! I'll validate shortly!

closed 2018-05-21 by FrancoisGrondin