introlab/odas #58: increase the gain

taospartan, 2018-05-20:
is it possible to increase the audio gain in a given search space using odas, **before** the audio gets processed? 

many thanks 

tao

FrancoisGrondin, 2018-05-21:
So if I understand, you would like to increase the gain only in a specific direction?

taospartan, 2018-05-21:
Yes please

FrancoisGrondin, 2018-05-21:
I see; I haven't implemented this yet. Is this something you'll find useful? I'm curious to see what's the reason why the gain would have to change for a given direction?

taospartan, 2018-05-21:
I’m trying to implement the recording of wildlife sounds by pointing the array at the animal, it would be of great use to be some distance from the animal being recorded. By increasing the gain in the direction of the animal this may be achieved. I hope:) but you’ll probably suggest a better way!

Many thanks:)

FrancoisGrondin, 2018-05-21:
Oh I understand. Ok in this case you can use the separated stream from the separation module. If you specify the target direction, you'll get a boost of gain in that direction.

taospartan, 2018-05-21:
Excellent many thanks:)

closed 2018-05-22 by FrancoisGrondin