introlab/odas #63: Meaning of level in SSL.

luc2yj, 2018-05-23:
What does level mean in SSL? e.g., nlevel, level, refined level, etc.

FrancoisGrondin, 2018-05-23:
Let's consider 3 variables: nLevels, level, and nRefinedLevels.

Variable **level** means the number of discretization steps of the search space for localization. The number of points on the full sphere corresponds to N = 10 * 4 ^ L + 2, where L is the number of levels.

L = 0, N = 12
L = 1, N = 42
L = 2, N = 162
L = 3, N = 642
L = 4, N = 2562
...

The variable **nLevels** represents the number of different resolution levels. For instance, we often have nLevels = 2, and L = 2 and L = 4. This means a first coarse scan is performed on a sphere with a resolution L = 2, and then for the most probable region of sound source activity, a fine scan is done in a sphere with L = 4 for more resolution. Having multiple levels allows to reduce computations instead of scanning entirely the full resolution sphere.

Finally, **nRefinedLevel** is used in the initialization step to discretize the continuous space around each point to optimize the size of the sliding window. My advice is to leave this parameter to 1.

luc2yj, 2018-05-23:
Thank you. I will take a deeper look into it.

closed 2018-05-23 by luc2yj

Femtofirst, 2023-07-12:
> Let's consider 3 variables: nLevels, level, and nRefinedLevels.
>
> Variable **level** means the number of discretization steps of the search space for localization. The number of points on the full [...]

Hi, great explanation! Could you also help with giving a bit explanation on this: I am using a minidsp 8-channel microphone for running ODAS, and below is some raw output from terminal - I know the first part is the result of SSL and the second part is the result of SST, the 'x,y,z ' in SSL result indicates the 3d vector of sound sources in the space and the 'E' stands for energy, indicating the intensity of sound.

{
"timeStamp": 2351,
"src": [
{ "x": -0.500, "y": 0.162, "z": 0.851, "E": 0.268 },
{ "x": -0.564, "y": -0.097, "z": 0.820, "E": 0.159 },
{ "x": -0.199, "y": 0.274, "z": 0.941, "E": 0.131 },
{ "x": -0.906, "y": 0.420, "z": 0.059, "E": 0.105 }
]
}
{
"timeStamp": 2351,
"src": [
{ "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
{ "id": 34, "tag": "dynamic", "x": -0.458, "y": -0.807, "z": 0.374, "activity": 0.000 },
{ "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 },
{ "id": 0, "tag": "", "x": 0.000, "y": 0.000, "z": 0.000, "activity": 0.000 }
]
}

And I am wondering:
    What do the "x,y,z" , the "ID" tag and the 'activity' in the SST result mean?
    Why there are only 4 lines of output for each?
    Which sound source is tracked in the result of SST? Is it the one with the highest energy?