introlab/odas #65: samplerate.mu and samplerate.sigma2

luc2yj, 2018-05-23:
What does the below two values mean?
    - samplerate.mu
    - samplerate.sigma2

FrancoisGrondin, 2018-05-24:
Ok, so you have the sample rate that is determined by your sound card, let's say 16 kSamples/sec. However, in real life, and depending on the sound card, this sample rate may slighty differ (ex: you could have 15.998 kSamples/sec). We need the sample rate in some module such as localization, as the sound wave propagation time delay in secs between two microphones is equal to (distance between mics) / sample rate. Since there might be a slight difference between theoretical and experimental values, we model the sample rate as a random variable, with mean given by "mu", and variance given by "sigma2". In other words, if I set the following:

mu = 16000
sigma2 = 1

It means I have a random variable x ~ N(16000,1) that models the sample rate. We propagate this uncertainty during the computations to ensure we round up properly and recover any disparities between the theoretical and experimental values of the sample rate.

Does this make sense?

Cheers

luc2yj, 2018-05-24:
Yes. It does make sense. Thank you.

closed 2018-05-24 by luc2yj