introlab/odas #67: How does plotting work?

alexengbloom, 2018-05-30:
I am currently modifying mod_ssl_process in mod_ssl.c. Even if I comment out the ssl calculation part I see a lot of points being plotted on the ODAS WEB as shown in the images attached. I think it is randomly generated as it changes whenever I restart odascore. Also, once generated it does not change its value. Could you explain why this is happening so that I can understand the issue? 

Thanks!

[image]
[image]
[image]
[image]

FrancoisGrondin, 2018-05-31:
It seams that as you are commenting the SSL processing, the system does not compute any beamforming energy on the sphere. It then tries to find the maximum values on non-initialized arrays, which can lead to this random result.

luc2yj, 2018-06-01:
Thank you. I have fixed the problem!

alexengbloom, 2018-06-01:
The problem can be fixed by commenting out `obj->nPots = 0;` from function `pots_zero(pots_obj * obj)`.

closed 2018-06-01 by alexengbloom