introlab/odas #68: Compatibility on MacOS (ALSA package only on linux)

yesheysamang, 2018-06-02:
Is ODAS available on MacOS yet?  I have been trying to build the library but the alsa package is not on macOS.  Thanks.

FrancoisGrondin, 2018-06-05:
I know this is an issue. I'm currently working on a "server" version of odas, where the client will be able to send via sockets raw signals and retrieve tracked directions and separated streams. By doing so, you could use any client (by example a Python one) and use audio drivers different from ALSA for macOS or Windows.

jan4984, 2018-06-05:
maybe portaudio is a choice.

FrancoisGrondin, 2018-06-05:
@jan4984 good point! Will have a look!

yesheysamang, 2018-06-06:
@jan4984 <https://github.com/jan4984> thanks will try that, but I think I'm
just better off running ubuntu on a virtual box.  What do you guys think?

FrancoisGrondin, 2018-06-27:
The new version of ODAS, which I will release in a few days, uses only sockets. You can thus pipe the stream from the soundcard to the socket using arecord or any other software, which therefore removes the dependency on ALSA.

mgw-sbic, 2020-05-20:
Hi all, seeing this old thread. We are looking for a way to take the sound from our video and write it into a MacOS loopback or audio hijack virtual sound device. FFmpeg supports reading from audio devices on MacOS, but not writing to audio devices. Is there a solution? THanks!

barracuda156, 2024-01-20:
> @jan4984 good point! Will have a look!

PortAudio and PulseAudio are supported on macOS, more so back to rather old versions (confirmed to work on 10.6, for example).