introlab/odas #70: Some question about mode of sss.

huotuichang1, 2018-06-07:
Excuse me,
I see there is mvdr in the sss_mode but there is not any code.why?Is it not effective?

FrancoisGrondin, 2018-06-09:
It is not implemented yet. MVDR uses a lot of computing resources, and was not real time on a raspberry pi. Is this something you'd need?

huotuichang1, 2018-06-10:
you are great~thank you~

taospartan, 2018-06-10:
What does MVDR stand for?

Sent from my iPhone

> On 9 Jun 2018, at 22:37, François Grondin <notifications@github.com> wrote:
> 
> MVDR

FrancoisGrondin, 2018-06-11:
A MVDR is good to eliminate "equally" all sounds coming from other directions than the desired one. From my limited experience with MVDR, here are the two drawbacks I observed:

1) When you have strong reflections generated by the desired source, these are considered as undesirable interference, and the MVDR tries to attenuate frequency bands that shouldn't be attenuated as they contain the target signal information. 
2) You need to compute and invert the correlation matrix at each frequency band, which requires significant computing resources

closed 2018-06-27 by huotuichang1