introlab/odas #74: Error when launch threads

chad1023, 2018-06-23:
We tried to do is start ODAS_CORE with Respeaker microphone array from the command line:
`bin/odascore -c config/test2.cfg -v`
and the core stopped with the error like this

[image]

The config file is correct because we can start the ODAS_WEB with the config successfully.

chad1023, 2018-06-24:
The config file we use:
```
# Configuration file for ReSpeaker circular sound card

version = "2.1";

# Raw

raw: 
{

    fS = 16000;
    hopSize = 128;
    nBits = 16;
    nChannels = 8; 

    # Input with raw signal from microphones
    interface: {
        type = "soundcard";
        card = 1;
        device = 0;
    }

}

# Mapping

mapping:
{

    map: (1,2,3,4,5,7);

}

# General

general:
{
    
    epsilon = 1E-20;

    size: 
    {
        hopSize = 128;
        frameSize = 256;
    };
    
    samplerate:
    {
        mu = 16000;
        sigma2 = 0.01;
    };

    speedofsound:
    {
        mu = 343.0;
        sigma2 = 25.0;
    };

    mics = (
        
        # Microphone 1
        { 
            mu = ( +0.0000, +0.0000, +0.0000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 90.0 );
        },

        # Microphone 2
[... 279 lines cut ...]

classify:
{
    
    frameSize = 1024;
    winSize = 3;
    tauMin = 32;
    tauMax = 200;
    deltaTauMax = 7;
    alpha = 0.3;
    gamma = 0.05;
    phiMin = 0.15;
    r0 = 0.2;    

    category: {

        format = "undefined";

        interface: {
            type = "blackhole";
        }

    }

}

taospartan, 2018-06-24:
You need to start odas_web first or instead of socket use interface terminal

Hope this helps

chad1023, 2018-06-25:
@taospartan 
Fixed it!
Thanks for your help !!!

closed 2018-06-25 by chad1023