introlab/odas #76: Regarding outline of SST (mod_sst_process).

alexengbloom, 2018-06-27:
I am currently trying to understand how mod_sst_process works in detail. Even after understanding how mod_sss.c and mod_ssl.c works, mod_sst_process is quite complicated and difficult to understand. It would extremely be helpful if someone could provide an outline of how mod_sst_process works.

Thanks!

-Alex

FrancoisGrondin, 2018-06-27:
Hi Alex,

I agree SST can be quite tricky. I just got a paper accepted, which details the SSL & SST processes used in ODAS. I need to make some minor revision, and then the final version will be available online.

For future reference, the title of this new paper is: **Lightweight and Opitmized Sound Source Localization and Tracking Methods for Open and Closed Microphone Array Configurations**

In the meantime, I can refer you to the previous paper where SST uses particle filtering: **The ManyEars open framework**

I hope this helps!

alexengbloom, 2018-06-28:
Does ODAS use the same flow as The ManyEars open framework? How about the parameters (especially the ones in configuration file)?

FrancoisGrondin, 2018-06-28:
Hi Alex,

Yes pretty much, except that ODAS now allows to use the particle filter, as with ManyEars, but also a Kalman filter, which is new. I will push some doc when I'm done with the major refactoring in the dev branch.

Cheers!

alexengbloom, 2018-07-03:
I have began to tweek the SST and have tried to add my own logic to it. Could you tell me _which part of the code handles the smaller highlighted dots?_ I know that the big highlighted dots are the result of SST. 

[image]

[image]

luc2yj, 2018-07-08:
@FrancoisGrondin Could you give me some clues on where to look at to deal with the small highlighted dots? Thank you!

luc2yj, 2018-07-08:
Hello @FrancoisGrondin . I also have the same question as @alexengbloom . Could you provide some details on where to look at?