introlab/odas #77: How recorded sound data pass from one mod to another?

tangsion, 2018-06-27:
I am reading odas source code these two days, and I am confused about how recorded sound data passing between mods. First, I know the sound data is recorded in src_hops_process, next the sound data is expected to pass to another module to apply algorithm， it seems to me the module is isolated.

Hoping your help. 
Thank you.

FrancoisGrondin, 2018-06-27:
Hi there!

I invite you to start looking at the dev branch, which will soon become the new version of ODAS. Now ODAS strictly uses socket, and the audio stream can be piped from the sound card through some tools (to change sample formats, and channels mapping) to the socket.

Let me know if you have questions,

tangsion, 2018-06-28:
Thank you for your reply！
I still not figure it out, maybe I made a confused  issue description. OK, I really want to know how the audio stream transmission between two modules, take, for example, raw source and raw connector, raw source and mapping module, mapping module and resample module, etc. There are so many modules and connectors in the code, I can‘t figure out the stream transmission， it drive me crazy.

Hoping for your help.
Thank you.

tangsion, 2018-06-28:
Add more details here.
Function snd_pcm_readi is call to record audio stream， and the audio stream is hold in src_hops_obj->buffer, I am stuck here， I can’t find any function/module/connector which is call to read the audio stream from src_hops_obj->buffer. It seems that the audio stream never transmit to the next step(like mapping or resample) to handle or apply algorithm.

FrancoisGrondin, 2018-06-28:
There is multiple thread that run in ODAS. There is a thread that calls the function `asrc_hops_thread`, and this function calls `src_hops_process`, which calls `src_hops_process_interface` at some point, and reads the buffer filled by the callback `snd_pcm_readi`.

Hopefully this is a bit more easy to follow in the new architecture currently in the dev branch.

tangsion, 2018-06-29:
Thank you for your help!
I have found out that some functions like **asrc_hops_connect** is the key.
OK, problem solved.

closed 2018-07-02 by FrancoisGrondin