introlab/odas #79: Correlate the source with the separation channel

moutaigua8183, 2018-06-29:
Hi, 

Assuming there are multiple sound sources, I am wondering if it is possible for ODAS to pair the source(direction) with the corresponding separation channel? If yes, would you be able to provide some potential solutions in details?

Thanks!

16thpolicy, 2018-06-29:
same issue here

FrancoisGrondin, 2018-06-30:
Hi there!

Yes it is possible! If you look at the order of tracked sources when you print them, you have this information. Suppose you can track up to 4 sources simultaneously, and you have two currently tracked sound sources, with id 6 and 9:

` id = 0, x = 0, y = 0, z = 0
id = 6, x = 0.56, y = 0.44, z = 0.33
id = 0, x = 0, y = 0, z = 0
id = 9, x = -0.4 y = 0.3, z = 0.8`

If you then look at the separated stream from ODAS, you have a 4 channels stream, and the second channel corresponds to source id = 6, and the fourth one to source id = 9. 

Note that with the new version (in the dev branch, and soon to be released), I've prepared a small tool that can write each separated source to a file with the corresponding id in the name, which makes everything much easier.

closed 2018-07-02 by FrancoisGrondin