introlab/odas #81: How do you differentiate between sounds coming from above and below the mic?

alexengbloom, 2018-07-03:
The program seems to run with a series of microphone that is placed on the same plane. As far as I understand such microphones can only cover half dome space and cannot differentiate sounds coming from below and above? However, it seems like ODAS does differentiate those sounds to some extent. How is this achieved?

FrancoisGrondin, 2018-07-03:
Can you provide me with you config file please?

alexengbloom, 2018-07-03:
It is the same configuration file you have uploaded for matrix_voice!

alexengbloom, 2018-07-09:
I have read through issue #19 . However, I still cannot understand how it works.
As mentioned in my first comment, I thought microphone arrays such as the matrix voice, which has all the mics on the same plane, can only cover half globe space; because the there are points that are exactly the same distance away from above and below the microphone array. I see that you consider 0,0,1 as 0 degree and give gain 1 from 0-80 degrees. However, how is it possible to differentiate 0,0,1 and 0,0,-1?

matrix voice: https://www.matrix.one/products/voice

FrancoisGrondin, 2018-07-09:
Ok I understand your point. In fact here, we *assume* the sound will come from front or back, and then limit the search space to a hemisphere in that direction. The system can however not *tell* the direction of arrival if you don't force it to be restricted to a specific direction. If you want to achieve full 3d localization, you need a 3d array.