introlab/odas #82: ODAS tracks only in one sector

Jansza, 2018-07-03:
Hi there! I'm sure this is a wrong channel for this but I'm having trouble with tracking using my custom microphone array. My array looks like this (pointy end is the mics direction):
[image]
It is a prototype and it working has been verified by just recording all the channels with Audacity.

So it is three mics with 45 degrees between them and one on top and a little back from the middle one.
I was using a tone generator and moved the mobile phone in front of the array. The problem is that ODAS seems to only track in the sector between the positive directions of all axes, so only in one sector.
Elevation tracking seemed to work better while azimuth stopped at some points and a separate source, that seemed to react to the movents of the sound source appeared.

 My best guess is still that it is still somehow a problem with my configuration, but I'm unsure what is wrong. I also have a question: what are the units used in mic position configuration?

```
mics = (        
        # Microphone 0
        { 
            mu = ( +0.400, -0.000, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( -1.000, +1.000, +0.000 );
            angle = ( 30.0, 150.0 );
        },
        # Microphone 1
        { 
            mu = ( +0.000, +0.200, +0.400 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +1.000, +0.000 );
            angle = ( 30.0, 150.0 );
        },
        # Microphone 2
        { 
            mu = ( +0.000, +0.400, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +1.000, +0.000 );
            angle = ( 30.0, 150.0 );
        },
        # Microphone 3
        { 
            mu = ( +0.400, +0.000, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +1.000, +1.000, +0.000 );
            angle = ( 30.0, 150.0 );
        }
    );
```

Other configuration is the same as the configuration tutorial in this wiki with the exception of sink configuration, which has been set up for ODAS web.

Best wishes,
Jaanus

taospartan, 2018-07-07:
I think your direction should be in a normalised vector format I.e. add up to 1

Hope this helps

Jansza, 2018-07-11:
Thanks for the hint, @taospartan.
I tinkered with the microphone location configuration a bit but the result didn't improve. It is now:

```
mics = (
        
        # Microphone 0
        { 
            mu = ( +0.120, -0.055, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( -0.500, +0.500, +0.000 );
            angle = ( 30.0, 150.0 );
        },

        # Microphone 1
        { 
            mu = ( +0.000, +0.070, +0.205 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +1.000, +0.000 );
            angle = ( 30.0, 150.0 );
        },

        # Microphone 2
        { 
            mu = ( +0.000, +0.145, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +1.000, +0.000 );
            angle = ( 30.0, 150.0 );
        },

        # Microphone 3
        { 
            mu = ( +0.120, +0.055, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.500, +0.500, +0.000 );
            angle = ( 30.0, 150.0 );
        }
    );

    spatialfilters: (

        {
            direction = ( +0.000, +1.000, +0.000 );
            angle = (80.0, 90.0);
        }

    );
```
The position mean seems to be metres, I measured my array and punched in the new numbers.
I consider the y-axis to be "forward" so I also tinkered with the spatial filter which, I hope now means that I'm looking for sources in a 90-degree arc in the y-direction.

The actual result is that the tracking works a bit worse now. :(
Might it be I'm reading my microphone [cardoid chart](https://static.bhphotovideo.com/lit_files/84878.pdf) wrong?

taospartan, 2018-07-11:
I would increase your angles on all mics to 90, 100 and on the spacial filter too

Then have a try again and let me know:)

Sent from my iPhone

> On 11 Jul 2018, at 10:30, Jansza <notifications@github.com> wrote:
> 
> 0.205 );

Jansza, 2018-07-13:
@taospartan, didn't seem to work. Might it be that my array is just unsuitable for this, might my microphones pick up the sound from a too wide angle?

Br,
Jaanus

taospartan, 2018-07-13:
If you want if you arecord some audio as raw and post it here along with the odas config and the distances between your mics I’ll have a look this weekend

Jansza, 2018-07-18:
[test-mic.zip](https://github.com/introlab/odas/files/2205705/test-mic.zip)
Sorry for the delay, @taospartan, I noticed your message on Monday's evening and didn't have the time yesterday. Here's the raw recording (arecord  hw:1,0 -d 10 -t wav -c 4 -f S32_LE -r 44100 /tmp/test-mic.wav).

I'll upload the picture of the array a bit later.

taospartan, 2018-07-21:
The audio needs to be type raw could you post raw audio then i’ll have a look?

Jansza, 2018-07-23:
Oh, my badSorry, @taospartan . Here are the correct ones (same parameters as previously but with raw type):
[speech.zip](https://github.com/introlab/odas/files/2218063/speech.zip)
[nowcorrect.zip](https://github.com/introlab/odas/files/2218064/nowcorrect.zip)

In the speech file I just stood in the middle and walker left and right. The other recording is a 500Hz tone, I moved my phone in a plus-sign pattern about a meter in front of the array.
```
mics = (
        
        # Microphone 0
        { 
            mu = ( -0.130, +0.055, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( -0.500, +0.500, +0.000 );
            angle = ( 90.0, 100.0 );
        },

        # Microphone 1
        { 
            mu = ( +0.000, +0.070, +0.205 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +1.000, +0.000 );
            angle = ( 90.0, 100.0 );
        },

        # Microphone 2
        { 
            mu = ( +0.000, +0.145, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +1.000, +0.000 );
            angle = ( 90.0, 100.0 );
        },

        # Microphone 3
        { 
            mu = ( +0.130, +0.055, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.500, +0.500, +0.000 );
            angle = ( 90.0, 100.0 );
        }
    );

    spatialfilters: (

        {
            direction = ( +0.000, +0.000, +1.000 );
            angle = (90.0, 100.0);
        }

    );
```

taospartan, 2018-07-30:
`mics = (
        
        # Microphone 1
        { 
            mu = ( -0.140, +0.055, -0.140 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( -0.500, +0.500, +0.000 );
            angle = ( 90.0, 100.0 );
        },

        # Microphone 2
        { 
            mu = ( +0.000, +0.070, +0.14 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +1.000, +0.000 );
            angle = ( 90.0, 100.0 );
        },

        # Microphone 3
        { 
            mu = ( +0.000, -0.18, -0.14 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +1.000, +0.000 );
            angle = ( 90.0, 100.0 );
        },

        # Microphone 4
        { 
            mu = ( +0.140, +0.055, -0.14 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.500, +0.500, +0.000 );
            angle = ( 90.0, 100.0 );
        }
    );

    spatialfilters: (

        {
            direction = ( +0.000, +1.000, +0.000 );
            angle = (90.0, 100.0);
        }

    );`

can you try this?

Jansza, 2018-09-16:
Sorry for the extreme delay, You've probably already forgotten the topic.
Moving the z-0 up to the middle of the array seems to have no effect, tracking is still chaotic for me.

Perhaps it is some of the other settings? It seems that the hopSize and sampling frequency seem to have an effect on tracking performance.

I now have the array much closer by and can try new configs faster if need be. I also tweaked the dimensions a bit, the idea stays the same - z axis 0 is raised to the middle of the array.

[scarlett_tao.txt](https://github.com/introlab/odas/files/2386723/scarlett_tao.txt)

I would try the new phoenix-branch, but it doesn't support the ODAS Web visualiser.

Br,
Jaanus