introlab/odas #83: Where can I find the detail specification of the 16SoundsUSB?

TooSchoolForCool, 2018-07-03:
Hi there,

Our lab bought a [16SoundsUSB](https://github.com/introlab/16SoundsUSB) from your lab recently, and we are trying to [odas](https://github.com/introlab/odas) on it. 

The problem is that we do not know how to set the configuration file appropriately. I went through the configuration tutorial [here](https://github.com/introlab/odas/wiki/Configuration), but I still have a hard time figuring out appropriate parameters.

For the raw data config, it gives an example as following,
```
raw: 
{

    fS = 44100;
    hopSize = 512;
    nBits = 16;
    nChannels = 8; 

    interface: {
        type = "file";
        path = "mics.raw";      
    };

};
```
I was wondering, what is the hopSize and nBits for the 16SoundsUSB, and how to set the Sample rate (fS) for the 16SoundsUSB (for example, I want to have the 16SoundsUSB with 16kHz, what should I do, do I need to update the firmware?).

FrancoisGrondin, 2018-07-06:
Hi!

Thank you for your interest in ODAS and the 16SoundsUSB hardware. To answer your questions:

1) 16SoundsUSB supports 44100 samples/sec for the sampling rate, so you should make the acquisition at that rate. However, in the config file, you have the parameter general.mu.samplerate to 16000, and odas will automatically handle resampling :)
2) hopSize can be set at 512 for raw.hopSize
3) nChannels should be set to 16
4) The number of bits (nBits) should be set to 32

Let me know if this answers your questions!

Cheers

TooSchoolForCool, 2018-07-06:
Thank you so much, your answer is really helpful.

So far, I still have two questions.

1. What do you mean by *resampling* in your first answer. If it is hard to explain, could you please tell me where does odas implement the resampling function, so that I can read the code directly :).
2. ''The number of bits (nBits) should be set to 32''. What if I set nBits to 16? Although it may lost some information, It should be okay right?

FrancoisGrondin, 2018-07-06:
1) Resampling, when you go from a high to low sample rates, with anti-aliasing filtering + downsampling, and when going from low to high sample rates by upsampling + interpolation filtering. In my case, since I have the STFT representation of signals, I do some workaround with that as well. The code is here: https://github.com/introlab/odas/blob/master/src/module/mod_resample.c
2) For now the firmware in 16SoundsUSB is configured to provide you with 32 bits number. I don't think the ALSA driver will let you start recording at 16 bits. Once ODAS gets the signal information, it is converted to float, and then you don't have to worry about fixed precision anymore.

TooSchoolForCool, 2018-07-06:
Perfect, I really appreciate it!

BTW, I tested the 16SoundsUSB with 16 bits config in [PyAudio](https://people.csail.mit.edu/hubert/pyaudio/), it automatically convert it to 32-bit float type when I recorded the sound as a .wav file. Finally, I figure out the reason.

Again, thank you soooooo much :).

TooSchoolForCool, 2018-07-09:
Hi there,

I reopened this issue since I found out that I still do not fully understand your magic...

As you mentioned above, 

> 16SoundsUSB supports **44100 samples/sec** for the sampling rate, so you should make the acquisition at that rate. However, in the config file, you have the parameter general.mu.samplerate to **16000**, and odas will automatically handle resampling

I was wondering why don't we just set the general.mu.samplerate to 44100? Is there any benefit if we set the general.mu.samplerate to 16000?