introlab/odas #85: Dds vs dgss

taospartan, 2018-07-10:
I’m using a single static target of x = 0.0, y= 1.0, z = 0, i’m Finding that dds far exceeds using dgss in the source separation beamforming. Dds seems to really enhance the targeted sound, dgss doesn’t seem to make a difference. Should this be the case? The sound source is 22meters away and quite weak. Do you need to use dynamic to make dgss work?
 Any help would be appreciated 

Cheers 

Tao

taospartan, 2018-07-24:
Any chance of some guidance on this I’m very puzzled

Many thanks

FrancoisGrondin, 2018-07-24:
Sorry, I didn't see the post! DDS is a delay and sum beamformer, and DGSS is a Geometric Source Separation beamformer. If you only have one active source, both DDS and DGSS should technically be similar... are you tracking more than a single source at a time?

taospartan, 2018-07-24:
No I’m only setting 1 target, but the sep raw’s are totally different, dds acts as would be assumed but dgss seems to make no difference to the audio:(