introlab/odas #86: How to disable spacial filter when using a microphone array that has mics on a single plane.

alexengbloom, 2018-07-10:
How to disable spacial filter when using a microphone array that has mics on a single plane.
Would modifying as below work?

    spatialfilters = (

        {

            direction = ( +0.000, +0.000, +1.000 );
            # angle = (80.0, 90.0);
            angle = (90.0, 90.0);

        }

    );

taospartan, 2018-07-10:
Francois drew this to show how the spatial filter worked

Which mic array are you using? Are the mics in a linear array?

alexengbloom, 2018-07-10:
It is in a circular array. Yeah. I am aware of that picture. My question is can we disable the filter?

taospartan, 2018-07-10:
I’ll be honest I’m confused are you just using static targets? The special filter is only there to restrict the search space for audio sources. If you aren’t using the dynamic source tracking then it won’t make any difference. 

If you explained what you are overall trying to achieve I may be able to help.

alexengbloom, 2018-07-10:
I am using dynamic source tracking. So the source could be on the same azimuth as the microphone array. That is why I want to avoid using the spacial filter.

alexengbloom, 2018-07-12:
@FrancoisGrondin Is there a way to disable the spacial filter? It seems like applying the spacial filter would filter out (or give less gain) some signals coming from the same height of the microphone array.

FrancoisGrondin, 2018-07-12:
If you set `angle = (180.0, 180.0);`, you will cover the full region with a gain of 1 everywhere.

If you have all mics on a single 2d plane and want to capture signals coming from the microphone array plane (at angle = 90), I suggest you do something like: `angle = (90,100)`