introlab/odas #88: I might have found a bug? Could you check?

luc2yj, 2018-07-12:
1) Order of the code in pots_zero.
```
    void pots_zero(pots_obj * obj) {
      obj->nPots = 0;
      memset(obj->array, 0x00, sizeof(float) * 4 * obj->nPots);
    }
```
isn't this supposed to be like
```
    void pots_zero(pots_obj * obj) {      
      memset(obj->array, 0x00, sizeof(float) * 4 * obj->nPots);
      obj->nPots = 0;
    }
```

2) Garbage value may get in when copying pots.
```
    void pots_copy(pots_obj * dest, const pots_obj * src) {

      dest->nPots = src->nPots;

      memcpy(dest->array, src->array, sizeof(float) * 4 * src->nPots);

    }
```
I tried changing it to:
```
    void pots_copy(pots_obj * dest, const pots_obj * src) {

      dest->nPots = src->nPots;
      memcpy(dest->array, src->array, sizeof(float) * 4 * src->nPots);
      memset(dest->array+(sizeof(float) * 4 * src->nPots), 0x00, sizeof(float) * 4 * (src->cfg_nPots-src->nPots));
      //when nPots is less than 4, the remainders in memory should be set in 0x00 to avoid to be set in random value.
      //src->cfg_nPots: I save the number of pots configured in the config file.

    }
```

FrancoisGrondin, 2018-07-12:
You're right! nPots should never bet set to 0. I just fixed it in the phoenix branch. Thank you for your feedback!

luc2yj, 2018-07-12:
@FrancoisGrondin From the pots_construct_zero function, i see that nPots are also initialized to the number of nPots set in the configuration file from the beginning (instead of 0). Does this mean that the value of nPots is always set to the number of nPots set in the configuration file?

FrancoisGrondin, 2018-07-12:
Yes, nPots for instance is set to 4. When we set the pots to zero, we actually clear the fields, x,y,z,E for each one of the nPots=4 potential sources.

luc2yj, 2018-07-12:
Okay. Seems like it's not necessary to do the second one then as the nPots are always the value defined in the confuguration file.

closed 2018-07-12 by luc2yj