introlab/odas #91: Connection closed after running odas

zengjiutian, 2018-07-15:
Description:
odas_web started by "npm start" in laptop and odas started by "./odaslive -c matrix_voice.cfg" on pi, i can see the new client connection in terminal of laptop. But soon later the connection closed. When i use "./odaslive -c matrix_voice.cfg -v" , the same problem shows. 
Could you please help me solve this problem ?
Thanks !
[image]
[image]

￼

closed 2018-07-26 by zengjiutian

albertmend, 2019-06-23:
Did you find the problem?

solodezaldivar, 2023-01-19:
Would be interested in a solution if anyone found one!

skainen, 2023-07-14:
Still no solution?