introlab/odas #96: System hang when running phoenix branch with SST and SSS on PI with Matrix Creator

elipschutz, 2018-07-27:
I appreciate the work you've put into this, it's significantly better than the doa provided by the Matrix Creator creators.

I was trying to get tracked sources and separated audio from the master branch, but it kept randomly hanging my PI. I switched to the phoenix branch and am able to get sst data for extended periods of time without issue. When I try to enable the "seps" port (assuming this is the correct port for separated audio) and run netcat on just the "tracks" port it hangs the system forcing me to pull the plug.

I'm aware that this could be a system configuration issue on my part, but is there anything I can tweak with odas that might fix this?

Thanks.

elipschutz, 2018-07-31:
Ah, after some testing with just netcat, it looks like my problem was that I started streaming audio in before connecting my client. Is there a modification I can make that will have odas discard data if no client is consuming it?

FrancoisGrondin, 2018-08-01:
Oh I see; hmm for now ODAS will try to buffer all data that comes in even if there is no client that consumes it.

closed 2018-09-10 by elipschutz