introlab/odas #97: Phoenix Branch Question

taospartan, 2018-07-30:
Using the Phoenix branch, is it possible to send your own target sources via a socket to odas in order to control the position of the beam manually? If so could you give me some guidance ?

Many thanks:)

FrancoisGrondin, 2018-07-31:
Yes it is :) 

You can send a JSON message with this format:

```
{
    "targets": 
    [
        { "tag": "your_first_target", "x": 1.0, "y": 0.0, "z": 0.0 },
        { "tag": "your_second_target", "x": 0.0, "y": 1.0, "z": 0.0 }
    ]
}
```

And then you forward it to the port associated to the target (ex: 5002) in the socket configuration file:

```
{
    "sockets":
    {
        "raw": 5001,
        "targets": 5002,
        
        "pots": 0,
        "tracks": 6002,
        "seps": 0,
        "pfs": 0,
        "iseps": 0,
        "ipfs": 0       
    }
}
```

Suppose your JSON message is in "targets.json" then you can for instance do something like this:

cat targets.json | netcat localhost 5002

Of course you can write your own script in any language that generates a JSON string and send it to ODAS asynchronously. You can thus refresh the target source over time.

Does this help?

taospartan, 2018-07-31:
Exceptionally clear and concise, exactly what I need 

Many thanks:)

taospartan, 2018-08-01:
Sorry to bother you but what’s the difference between seps and iseps? Can the sep raw settings in the same method as targets? E.g. gain?

FrancoisGrondin, 2018-08-02:
iSep is a stream that uses additional channels to carry tracking information. This way you can have all info in sync and send it through pipes to another process ;)

taospartan, 2018-08-02:
Is it then possible to adjust the gain via the socket?

FrancoisGrondin, 2018-08-02:
The gain of the separated signal?

taospartan, 2018-08-02:
Yes:)

FrancoisGrondin, 2018-08-02:
You mean having a parameter you can send async that will change the volume?

taospartan, 2018-08-02:
Yes that would be awesome !

FrancoisGrondin, 2018-08-03:
I see! For now this is nothing like that, but I'll keep this in mind for future feature ;)