introlab/odas #98: SST threshold for showing up as a source

dextragon, 2018-07-31:
Hey,
first of all thanks for this amazing library, I've played around with it a little and its really impressive!
I'm using a Matrix Creator, and what I noticed is that for Sound Source Localization I can speak pretty far away from the array/pretty quietly and still show up as a sound source.
However, when I use Sound Source Tracking, I have to get a lot closer/speak louder to be recognized as a source. I am not very familiar with the inner workings of the algorithms, so I'm not sure what most of the parameters in the config file do (in the wiki it just says to leave them at what they are :D). Is there some parameter I can change to lower the threshold/increase the sensitivity of the Sound Source Tracking, so that quieter sources will also show up?

Oh and one other thing (its not really an issue, so I didn't want to open one for it, just something maybe worth noting): I've noticed that the output of the Sound Source Separation (when using signed 16bit integers) is not fully centered, the mean lies roughthly around -3000 for me, if there is a signal present (if there is no signal its on 0). (Maybe this is simply due to my array missing one mic, which I also adjusted in the config file.)

FrancoisGrondin, 2018-08-01:
Hi there!

For the sound source detection threshold, I think the issue #28 may answer your question.

For the offset after separation, this is strange. Have you tried setting the bin 0 in the freq frames to zero? That is equivalent to removing any DC offset.

Cheers!

dextragon, 2018-08-02:
Thanks for your quick reply!
Issue [#28](https://github.com/introlab/odas/issues/28) was exactly what I was lookign for :)

For setting the bin 0 to zero, how would I do that? The output I'm getting (to the socket I specified in the config file) seems to be in the time domain. Is this a change I need to make in the odas code?
I am using the separated (not the postfiltered) output, if that makes a difference.

FrancoisGrondin, 2018-08-02:
Ok, try to add these two lines of code

```
obj->out1->freqs->array[0] = 0.0f;
obj->out1->freqs->array[1] = 0.0f;
```

At this place: https://github.com/introlab/odas/blob/cd6658e3983b2bc5f5db30b64d6af49988383259/src/module/mod_sss.c#L572

And let me know if it solves your issue!

dextragon, 2018-08-13:
Hey, sorry for the late reply.
I figured out the cause for the offset: It was the missing microphone. I thpught I had compensated for this in the config file, but appearantly I removed the wrong mic 😅 
Now that it is replaced everything is fine again :)

closed 2018-08-13 by dextragon