introlab/odas #99: General instructions to read direction and angle

pkalebu, 2018-08-01:
I'm unsure how to get started with the library. I have it installed as per the wiki, but don't know how to read out the localization data without using the visualization tool. I am using a ReSpeaker USB 4-mic array. Does anyone know how I can go about reading out the raw localization values?

FrancoisGrondin, 2018-08-01:
You can print the potential/tracked sources in the terminal, is that what you are looking for?

FrancoisGrondin, 2018-08-02:
You can change the config file like this:

```
    # Output to export tracked sources
    tracked: {

        format = "json";

        interface: {
            type = "terminal";
        };
    };
```

pkalebu, 2018-08-02:
Awesome, thanks!

closed 2018-08-02 by pkalebu