introlab/odas #103: How to configure gain properly for working in long distance?

jamesshao8, 2018-08-08:
We are trying to receive signals from a longer distance, like 10 meters.

We experimented with audacity and found that a single channel can hear normal speech at around 4 meters. But with odas we can only hear at around 2~3 meters. I think with beamforming, odas should be able to provide longer distance of receiving than single channel microphone. So we wonder if it's a setting problem.

We saw in the cfg file there are various gain, like gain_sep gain_pf and another gain in postfiltered section. Which gain matters the most and what is the largest value can we put in there? 

Thanks!

taospartan, 2018-08-08:
Which mic array are you using?

jamesshao8, 2018-08-08:
@taospartan  I am using 16SoundsUSB and the electret version of microphone board. I made them locally according to the design file from introlab. The sensitivity of the electret microphone board is -44dB. And first stage of amplification is around 20 times, which is strange for me because according to the datasheet it should be 40dB (100 times).

taospartan, 2018-08-12:
I’d be looking at the dynamic range of the microphones. With dss beamforming I think I’ve gone up to around sep_gain 50. I’d experiment with static beamforming, giving odas your own targets. Also I’ve found having the array in a horizontal orientation makes a difference.

jamesshao8, 2018-08-13:
@taospartan I am using post-filtered instead of separate. But once increased to 10, there is obvious distortion. We have already tried static beamforming, by setting the cfg file. But I do notice there is a phoenix branch, is it optimized for this kind of work?

taospartan, 2018-08-14:
> 24k 32k 44.1k is the sampling rate right? Are you sure that sampling rate would affect the distance? The ADC bit size has something to do with the dynamic range. I am using the 16SoundsUSB with 32bit, and trying to increase the signal in the analog part. @taospartan

I’ll be honest I’m not sure about anything but trying different ideas and seeing what works!:)

jamesshao8, 2018-08-14:
In RF world, oversampling is used to solve aliasing. This indeed reminds me of a problem that I am using a sample rate of 16kHz to sample from my electret mic with a maximum frequency of 20kHz. I will try to setup the cfg file to sample at higher frequency or choosing a microphone with lower frequency.

taospartan, 2018-08-18:
@jamesshao8 have you had any success?

jamesshao8, 2018-08-19:
Hello taospartan,

We have used electret mic in other projects before. And the recording distance should be way larger than what we are doing with odas and 16soundsusb. We are suspecting that this is not a hardware problem. Either the problem is with odas, or we are just not tuning to a enough volume.

We are developing our own software to do beamforming and real-time listening to the audio. And we found maybe it's not the problem of pf_gain. Because if we increase pf_gain, distortion is created.  So instead, it's a problem when converting float to short, like multiplying 0x7FFF for 16 bit samples. But when we multiply this constant, or any constant like a volume, there is another kind of distortion. However, when we put the recorded output to gnuradio, there is no distortion when changing volume. We are still looking if it's a coding problem.

taospartan, 2018-08-20:
What kind of distances are you working at? I’ve been using sox in the last few days to increase the gain of the pf output but my hunch is to try and increase the gain on the input before odas and see if that can effect the distance, but this requires more resources than a humble raspberry pi 3b with 16 mics! Also I’m looking at the noise floor of the mics to make sure signals aren’t being masked

jamesshao8, 2018-08-21:
I am now able to receive voice signal from 3 meters. My goal is 10 meters. What is sox? I am also making mems microphones for 16SoundsUSB, but I will get those boards at the end of this week. Could you explain why computing power effects the receiving distance? I am using a laptop to run odas, so computing power isn't a problem. Once I have confirmed that mems version is better than electret mics, I will use my own mics with even higher sensitivity and SNR.

jamesshao8, 2018-08-25:
@taospartan I have tried the mems mic design for the 16SoundUSB. They can provide me with higher gain indeed. And my own modification of the amplification loop increase the distance further. But it's still not enough for me. Other than that, I found when in a close distance, the beamforming effect is obvious, in the sense that, the sound level in wanted direction is much higher than from other direction. But when the distance go up, the sound level becomes similar in all directions.

taospartan, 2018-08-25:
What distances does it drop off ? And what’s the noise from of the mics?

jamesshao8, 2018-08-25:
I can hear voice from around 7 meters. But there is no difference whether I hit right on the beamforming spot. I think to have obvious beamforming effect, I should come near to around 3 meters. I am suspecting that if there is a threshold for the beamforming module in odas to take effect?

taospartan, 2018-08-25:
Or the noise floor of the mics is affecting it, just out of interest what is the layout of your mics?

jamesshao8, 2018-08-25:
Yes maybe it's caused by the noise floor, which cannot be distinguished by direction anymore.

The mic I am using is designed by introlab.
You can find them here:
https://github.com/introlab/xSoundsMicrophones

taospartan, 2018-08-25:
Are your mics in a horizontal or vertical plane? 

I did have a thought the increasing the bit rate may have an effect but would probably need lots more resources. The theory being the dynamic range would be increased, what do you think?

jamesshao8, 2018-08-25:
The mics are facing towards the target, so most of the time we put it vertically. We use 32bit in the hardware configuration but then decreased to 16 bit in pf data. We did that because it's easier to process the output by webrtc noise suppression algorithm.

taospartan, 2018-08-25:
I know this will sound odd but try it horizontally, that’s what I do with the uma16. I have also used sox to increase the gain before the audio reaches odas. I can hear voices clearly just using odas pf doing this. Also see if the beamforming works in the sep stream, I use the spectrograph view in audacity to check

jamesshao8, 2018-08-27:
@taospartan  I tried to place the mics horizontally, but unfortunately it's not getting better. However, that does reminds me of trying different shapes of mic array. I will try UCA with different diameters, or even a plane matrix.

taospartan, 2018-08-27:
Good luck, the spacing may be a factor too