introlab/odas #104: Using ReSpeaker Mic Array v2.0, CPU Usage ~95%-100%

ericg689, 2018-08-09:
I'm a high school student running an experiment on sound localization. My setup uses a ReSpeaker Mic Array v2.0 connected to a Raspberry Pi 3 Model B. I'm experiencing an issue where the CPU usage is usually between 95% and 100%. The application usually freezes when I try launching it. I can still move my mouse, and I can see the CPU Usage and CPU Temp. change from time to time. However, the Source Elevation and Source Azimut usually don't show anything. It's very slow to respond to stimuli. I have the config file as the odas.cfg that was originally in the ReSpeaker Mic Array library. 

I followed the installation process, but skipped the configuration section. Could this be the source of the problem? I was also wondering if I could use ODAS on an Asus Tinkerboard. Thanks in advance!

taospartan, 2018-08-09:
Hi there, from my experience running odas and odas_web on the same machine ( especially a pi) doesn’t offer the best experience. odas_web is quite resource intensive as an interface.
 As the data could be collected from odas’s sockets in json maybe you could limit what data you need and make a little web interface? 

It all depends on what you are trying to do?

Tao

ericg689, 2018-08-09:
Thanks for your response! If you say that running odas and odas_web on the same machine won't work, what solution would you recommend? I'm just trying to get the program running first. I'm not sure about the steps after that, but I probably want to see if I can record the data into a file.

taospartan, 2018-08-09:
First you need to edit the respeaker config you are using, changing the type from socket to file and path. Like this

interface: {
type = "file";
path = "/temp/filename.ext";
}

Then run odas from command line with this new cfg, the data will go into files you’ve described above

Tao

ericg689, 2018-08-09:
Should I be using the ReSpeaker config file or the odas.cfg file that originally comes with the ReSpeaker library? Also, which of the "interface:{ ..." should I be changing? I see that there are several, each under a different section. Will this reduce CPU usage? I'm currently just trying to run the program and not collect data yet. Thank you!

taospartan, 2018-08-09:
The one that comes with the respeaker 

All the interfaces have differing purposes, if you are just interested in the sound source tracking  set the interface in that section.

I would set the other interfaces that you aren’t using to this:

format = "undefined";

        interface: {
            type = "blackhole";
        }

If you want to save the audio files generated in separated and postfiltered just leave them as they are

Hope this helps

Tao

ericg689, 2018-08-09:
Thanks for the help so far! I'm trying to use information from the application to move servo motors towards a sound source. Should I be using SSL data or SST data? Thank you again!

ericg689, 2018-08-09:
Thanks! For some reason, I can't get the mic array to track sound sources even though it can detect sounds pretty well. Any solution for this? Attached is the config file I am using. I changed the parameters that you mentioned and I also changed the sensitivity so that the mic array is a bit more sensitive. Thanks again!

`# Configuration file for XMOS circular sound card

version = "2.1";

# Raw

raw: 
{

    fS = 16000;
    hopSize = 128;
    nBits = 16;
    nChannels = 6; 

    # Input with raw signal from microphones
    interface: {
        type = "soundcard";
        card = 1;
        device = 0;
    }

}

# Mapping

mapping:
{

    map: (2, 3, 4, 5);

}

# General

general:
{
    
    epsilon = 1E-20;

    size: 
    {
        hopSize = 128;
        frameSize = 256;
    };
    
    samplerate:
    {
        mu = 16000;
        sigma2 = 0.01;
    };

    speedofsound:
    {
        mu = 343.0;
        sigma2 = 25.0;
    };

    mics = (

        # Microphone 2
        { 
            mu = ( -0.032, +0.000, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 100.0 );
        },

        # Microphone 3
        { 
            mu = ( +0.000, -0.032, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 100.0 );
        },

        # Microphone 4
        { 
            mu = ( +0.032, +0.000, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 100.0 );
        },

        # Microphone 5
        { 
            mu = ( +0.000, +0.032, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 80.0, 100.0 );        
        }
        
    );

    # Spatial filter to include only a range of direction if required
    # (may be useful to remove false detections from the floor)
    spatialfilters = (

        {
            direction = ( +0.000, +0.000, +1.000 );
            angle = (80.0, 100.0);

        }    

    );

    nThetas = 181;
    gainMin = 0.25;

};

# Stationnary noise estimation

sne:
{
    
    b = 3;
    alphaS = 0.1;
    L = 150;
    delta = 3.0;
    alphaD = 0.1;

}

# Sound Source Localization

ssl:
{

    nPots = 4;
    nMatches = 10;
    probMin = 0.5;
    nRefinedLevels = 1;
    interpRate = 4;

    # Number of scans: level is the resolution of the sphere
    # and delta is the size of the maximum sliding window
    # (delta = -1 means the size is automatically computed)
    scans = (
        { level = 2; delta = -1; },
        { level = 4; delta = -1; }
    );

    # Output to export potential sources
    potential: {

        format = "undefined";
        # format = "json";

        interface: {
            type = "blackhole";
            # type = "socket"; ip = "127.0.0.1"; port = 9001;
        };

    };

};

# Sound Source Tracking

sst:
{  

    # Mode is either "kalman" or "particle"

    mode = "kalman";

    # Add is either "static" or "dynamic"

    add = "dynamic";    

    # Parameters used by both the Kalman and particle filter

    active = (
        { weight = 1.0; mu = 0.3; sigma2 = 0.0025 }
    );

    inactive = (
        { weight = 1.0; mu = 0.1; sigma2 = 0.0025 }
    );

    sigmaR2_prob = 0.0025;
    sigmaR2_active = 0.0225;
    sigmaR2_target = 0.0025;
    Pfalse = 0.1;
    Pnew = 0.1;
    Ptrack = 0.8;

    theta_new = 0.9;
    N_prob = 5;
    theta_prob = 0.8;
    N_inactive = ( 250, 250, 250, 250 );
    theta_inactive = 0.9;

    # Parameters used by the Kalman filter only

    kalman: {

        sigmaQ = 0.001;
        
    };
   
    # Parameters used by the particle filter only

    particle: {

        nParticles = 1000;
        st_alpha = 2.0;
        st_beta = 0.04;
        st_ratio = 0.5;
        ve_alpha = 0.05;
        ve_beta = 0.2;
        ve_ratio = 0.3;
        ac_alpha = 0.5;
        ac_beta = 0.2;
        ac_ratio = 0.2;
        Nmin = 0.7;

    };

    target: ();

    # Output to export tracked sources
    tracked: {

        format = "json";

        interface: {
            type = "file";
            path = "/home/linaro/odas/temp/tracks.txt";
            # type = "socket"; ip = "127.0.0.1"; port = 9000;
        };

    };

}

sss:
{
    
    # Mode is either "dds", "dgss" or "dmvdr"

    mode_sep = "dgss";
    mode_pf = "ms";

    gain_sep = 1.0;
    gain_pf = 10.0;

    dds: {

    };

    dgss: {

        mu = 0.01;
        lambda = 0.5;

    };

    dmvdr: {

    };

    ms: {

        alphaPmin = 0.07;
        eta = 0.5;
        alphaZ = 0.8;        
        thetaWin = 0.3;
        alphaWin = 0.3;
        maxAbsenceProb = 0.9;
        Gmin = 0.01;
        winSizeLocal = 3;
        winSizeGlobal = 23;
        winSizeFrame = 256;

    };

    ss: {

        Gmin = 0.01;
        Gmid = 0.9;
        Gslope = 10.0;

    };

    separated: {

        fS = 16000;
        hopSize = 128;
        nBits = 16;        

        interface: {
            type = "file";
            path = "separated.raw";
        };        

    };

    postfiltered: {

        fS = 16000;
        hopSize = 128;
        nBits = 16;        
        gain = 10.0;

        interface: {
            type = "file";
            path = "postfiltered.raw";
        };        

    };

};

classify:
{
    
    frameSize = 4096;
    winSize = 3;
    tauMin = 88;
    tauMax = 551;
    deltaTauMax = 20;
    alpha = 0.3;
    gamma = 0.05;
    phiMin = 0.5;
    r0 = 0.2;    

    category: {

        format = "undefined";

        interface: {
            type = "blackhole";
        }

    }

}
`

taospartan, 2018-08-09:
Can you post your tracks.txt?

ericg689, 2018-08-09:
Sure, here it is:
[tracks.txt](https://github.com/introlab/odas/files/2276029/tracks.txt)

ericg689, 2018-08-10:
I believe the 6 channel one.

ericg689, 2018-08-10:
After looking through the tracks.txt file, I noticed that sometimes there's a change from (0, 0, 0), but always reverts back to (0, 0, 0). Not sure if that means that the SST is working or not.

taospartan, 2018-08-10:
If it was me I’d record some audio using the respeaker and the app arecord. Make sure you capture it at 16bit 16k in raw format.

Then import the audio into audacity to check the audio is working and you have 6 channels. 

I think we need to confirm audio is working before we go further:)

ericg689, 2018-08-10:
I think the audio is working. I just tried using SSL and SST in tandem (commented out the file section for SST), and the dot for SST appeared. However, I then turned SSL off and put back the file section for SST, which made it so that there was nothing appearing on the interface (with the globe thing). There was data in the tracks.txt file though.

taospartan, 2018-08-10:
Hmmm puzzling I’d still look that the audio files first to prove a solid foundation I have a working config for my respeaker if you think that would help?

ericg689, 2018-08-10:
Sure, if I could see it I can compare it with mine

taospartan, 2018-08-10:
# Configuration file for respeaker 4 mic array v2.0
version = "2.1";

# Raw

raw: 
{

    fS = 16000;
    hopSize = 128;
    nBits = 16;
    nChannels = 6; 

    # Input with raw signal from microphones
    interface: {
        type = "file";
        path = "test.raw"
    }

}

# Mapping

mapping:
{

    map: (2, 3, 4, 5);

}

# General

general:
{
    
    epsilon = 1E-20;

    size: 
    {
        hopSize = 128;
        frameSize = 256;
    };
    
    samplerate:
    {
        mu = 16000;
        sigma2 = 0.01;
    };

    speedofsound:
    {
        mu = 343.0;
        sigma2 = 25.0;
    };

    mics = (

        # Microphone 2
        { 
            mu = ( +0.0239, +0.0239, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 90.0, 100.0 );
        },

        # Microphone 3
        { 
            mu = ( -0.0239, +0.0239, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 90.0, 100.0 );
        },

        # Microphone 4
        { 
            mu = ( -0.0239, -0.0239, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 90.0, 100.0 );
        },

        # Microphone 5
        { 
            mu = ( +0.0239, -0.0239, +0.000 ); 
            sigma2 = ( +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000, +0.000 );
            direction = ( +0.000, +0.000, +1.000 );
            angle = ( 90.0, 100.0 );        
        }
        
    );

    # Spatial filter to include only a range of direction if required
    # (may be useful to remove false detections from the floor)
    spatialfilters = (

        {
            direction = ( +0.000, +0.000, +1.000 );
            angle = (90.0, 100.0);

        }    

    );

    nThetas = 181;
    gainMin = 0.25;

};

# Stationnary noise estimation

sne:
{
    
    b = 3;
    alphaS = 0.1;
    L = 150;
    delta = 3.0;
    alphaD = 0.1;

}

# Sound Source Localization

ssl:
{

    nPots = 4;
    nMatches = 10;
    probMin = 0.5;
    nRefinedLevels = 1;
    interpRate = 4;

    # Number of scans: level is the resolution of the sphere
    # and delta is the size of the maximum sliding window
    # (delta = -1 means the size is automatically computed)
    scans = (
        { level = 2; delta = -1; },
        { level = 4; delta = -1; }
    );

    # Output to export potential sources
    potential: {

        format = "undefined";
        #format = "json";

        interface: {
             type = "blackhole";
            #type = "socket"; ip = "127.0.0.1"; port = 9001;
        };

    };

};

# Sound Source Tracking

sst:
{  

    # Mode is either "kalman" or "particle"

    mode = "kalman";

    # Add is either "static" or "dynamic"

    add = "dynamic";    

    # Parameters used by both the Kalman and particle filter

    active = (
        { weight = 1.0; mu = 0.4; sigma2 = 0.0025 }
    );

    inactive = (
        { weight = 1.0; mu = 0.25; sigma2 = 0.0025 }
    );

    sigmaR2_prob = 0.0025;
    sigmaR2_active = 0.0225;
    sigmaR2_target = 0.0025;
    Pfalse = 0.1;
    Pnew = 0.1;
    Ptrack = 0.8;

    theta_new = 0.9;
    N_prob = 5;
    theta_prob = 0.8;
    N_inactive = ( 250, 250, 250, 250 );
    theta_inactive = 0.9;

    # Parameters used by the Kalman filter only

    kalman: {

        sigmaQ = 0.001;
        
    };
   
    # Parameters used by the particle filter only

    particle: {

        nParticles = 1000;
        st_alpha = 2.0;
        st_beta = 0.04;
        st_ratio = 0.5;
        ve_alpha = 0.05;
        ve_beta = 0.2;
        ve_ratio = 0.3;
        ac_alpha = 0.5;
        ac_beta = 0.2;
        ac_ratio = 0.2;
        Nmin = 0.7;

    };

    target: (  );
   

    # Output to export tracked sources
    tracked: {

        format = "json";

        interface: {
             type = "file";
             path = "tracks.txt";
            #type = "socket"; ip = "127.0.0.1"; port = 9000;
        };

    };

}

sss:
{
    
    # Mode is either "dds", "dgss" or "dmvdr"

    mode_sep = "dds";
    mode_pf = "ss";

    gain_sep = 1.0;
    gain_pf = 10.0;

    dds: {

    };

    dgss: {

        mu = 0.01;
        lambda = 0.5;

    };

    dmvdr: {

    };

    ms: {

        alphaPmin = 0.07;
        eta = 0.5;
        alphaZ = 0.8;        
        thetaWin = 0.3;
        alphaWin = 0.3;
        maxAbsenceProb = 0.9;
        Gmin = 0.01;
        winSizeLocal = 3;
        winSizeGlobal = 23;
        winSizeFrame = 256;

    };

    ss: {

        Gmin = 0.01;
        Gmid = 0.9;
        Gslope = 10.0;

    };

    separated: {

        fS = 16000;
        hopSize = 128;
        nBits = 16;        

        interface: {
           #type = "blackhole";
           #type = "socket"; ip = "127.0.0.1"; port = 10000;
           type = "file"; path = "/tmp/sep.raw";

        };        

    };

    postfiltered: {

        fS = 16000;
        hopSize = 128;
        nBits = 16;        
        gain = 10.0;

        interface: {
            #type = "blackhole";
            #type = "socket"; ip = "127.0.0.1"; port = 10010;
            type = "file"; path = "/tmp/pf.raw";
        };        

    };

};

classify:
{
    
    frameSize = 4096;
    winSize = 3;
    tauMin = 88;
    tauMax = 551;
    deltaTauMax = 20;
    alpha = 0.3;
    gamma = 0.05;
    phiMin = 0.5;
    r0 = 0.2;    

    category: {

        format = "undefined";

        interface: {
            type = "blackhole";
        }

    }

}

ericg689, 2018-08-10:
Is there any reason for the mapping you used 0.0239 instead of 0.032 for the mu and 90 instead of 80 for the angle? Also, what do the abbreviations mean for SSS? I believe we have some differences there in the mode_sep and mode_pf. Besides those differences, our files are essentially the same. 

I believe you are also using the ReSpeaker Mic Array v2.0 right? Thanks so much for the help!

taospartan, 2018-08-10:
I rotated the coords to be exactly as on respeaker v2 as the makers have them lined up in a perfect horizontal vertical cross, if you draw them out you’ll see what I mean the other settings are what I’ve found to be best :)

No trouble at all:)

ericg689, 2018-08-10:
Gotcha. In that case, I will try to use your settings and see how it goes. I’ll let you know how it goes!

taospartan, 2018-08-10:
Cool:)

ericg689, 2018-08-13:
It seems like it's performing a lot better, but the tracking still doesn't show up on the GUI. I'm still able to get data though, so I think it should be fine. You mentioned in your very first response to this thread that running odas and odas_web on the same machine would lead to the slow speeds I experienced on the Pi. Is there a way to run them separately, that way alleviating some of the processing pressure on the Pi? 

Also, do you know if it's possible to get sound level readings from the ReSpeaker mic array?
Thanks!

taospartan, 2018-08-13:
Just run odas web separately on a laptop and use sockets to the ip of the laptop from the pi, that’s what I did

taospartan, 2018-08-13:
Sound readings... maybe you could use the activity field in the tracks json?

ericg689, 2018-08-13:
Regarding the sound readings, I thought the activity field was just the probability that the source that the program is tracking is "active." In other words, the probability whether that source is actually making noise or not. I may be getting this wrong, so please correct me if that is the case!

ericg689, 2018-08-13:
@taospartan I updated odas_web recently and now the tracking doesn't work at all. Have you experienced this yet?

ericg689, 2018-08-13:
I updated npm this morning (it gave me a message saying that I should type in "npm i -g npm"). However, after I updated this, the SST stopped working. I'm not able to obtain any data. I reinstalled the ODAS library and the ReSpeaker library, yet it still does not work. Not sure what the issue is. Any help on this?

taospartan, 2018-08-13:
I would post this issue in the odas web repo, it’s a bit out of my league

ericg689, 2018-08-13:
Ok, thanks!