introlab/odas #107: Matrix creator project

magic-dave, 2018-08-13:
Good afternoon,

The matrix creator demo works exactly as required, but I am trying to extract the raw angle prediction. I cannot seem to find where you are providing or calculating the direction of approach data.

Any advice you have on how I can get this data would be greatly appreciated.

Thank you,
Dave

closed 2018-08-27 by magic-dave

jake266637, 2018-11-22:
Did you end up finding a solution?

magic-dave, 2018-11-22:
Good morning,

Yes, this ticket can be closed. I had forgotten about this.

Thank you,

Dave

jake266637, 2018-11-22:
What was the solution?

magic-dave, 2018-11-22:
The angle is determined in increase_pots() using the X and Y coordinates of the sound source. It is in a variable called angle_xy - you can assign the value to a variable that is accessible anywhere. I had a variable called holder that was used to carry the angle data from the increase_pots() function elsewhere using ROS.

jake266637, 2018-11-22:
Oh great thanks