introlab/odas #112: strange audio recordings

kimchi88, 2018-08-23:
Dear Odas team,
I am playig around with your tool using a virtual machine running Ubuntu 18 and the last available version in GitHub of Odas and Odas-web.
My Microphone is a ReSpeaker Mic Array v2.0. Firmware is the 6 channel one.
My experiment is set up with the Respeaker in the middle and two persons talking independently from the opposite sides of the ReSpeaker at a distance of about 2 meters from the microphone (about 4 meters distance between the two people). 
The recording is pretty strange. here in attached you can find the recordings + my configuration file.
Any suggestion to improve?

Thanks for your help!

[odasmy.zip](https://github.com/introlab/odas/files/2312806/odasmy.zip)
[ODAS_test.zip](https://github.com/introlab/odas/files/2312808/ODAS_test.zip)

FrancoisGrondin, 2018-09-05:
Hi,

I listened to your samples. This is indeed quite strange. Have you listen to the raw audio recorded by each channel? Can you tell me if you get the same distortion?

kimchi88, 2018-09-06:
Hi,
Using Audacity it seems ok. Here in attached a six channels record.
[6ChRecTest.zip](https://github.com/introlab/odas/files/2355407/6ChRecTest.zip)

Maybe I am doing something wrong since I don't know exactly how to tune the recording in Odas-web. Could you give me more info about how the recording GUI works? Like when it actually starts to record? how it decides when to start and stop a single file? what does the "determine if separated audio is recorded" check-button does?  Any help is appreciated.
I really like your project and I would like to going on testing and using it.

Thanks!

FrancoisGrondin, 2018-09-08:
Hi,

Let me have @GodCed here for the GUI part. The RAW files seem indeed ok. Please try the following config for the separation and tell me if you still have the same artifacts:

`mode_sep = "dgss";`

Thanks

GodCed, 2018-09-08:
@kimchi88 I suggest you take a look at [this part of the ODAS Studio documentation](https://github.com/introlab/odas_web#record) for more information about the recording GUI.

`Like when it actually starts to record? how it decides when to start and stop a single file?`
It starts and stops according to ODAS tracking. When a tracked source appears, recording starts. When it disappears, recording stops. A file is created for each tracked sources, named after the timestamp, the ODAS source ID and the audio stream (postfiltered _pf or separated _sp).

`what does the "determine if separated audio is recorded" check-button does?`
It enables recording. When checked, recording starts when a source appears. Existing sources at the time of checking are not recorded. When unchecked, current recordings are stopped and no new one will start.

GodCed, 2018-09-08:
For easier debugging, I suggest you use ODAS to record directly your results to WAV files instead of trough ODAS Studio to help identify wether the issue comes from ODAS or ODAS Studio.

kimchi88, 2018-09-10:
@FrancoisGrondin using the setting you suggested the recording seems to be improved. Still, many pieces of record are distorted. Could this be related to the massive amount of CPU usage required by ODAS-Web? It is always close to 100%.

@GodCed if I use odas from the command line without have odas-web opened, I get the following error:

./odaslive -v -c /home/fse/usb_4_mic_array/odasmy.cfg 
+--------------------------------------------+
|    ODAS (Open embeddeD Audition System)    |
+--------------------------------------------+
| Author:  Francois Grondin                  |
| Email:   francois.grondin2@usherbrooke.ca  |
| Website: introlab.3it.usherbrooke.ca       |
| Version: 1.0                               |
+--------------------------------------------+
| + Initializing configurations...... [Done] |
| + Initializing objects............. [Done] |
| + Launch threads................... Sink pots: Cannot connect to server
Sink tracks: Cannot connect to server

If I try to launch the script ./odasserver doesn't do anything.
Is there any other way to launch odas without odas-web?

Thanks for the help

GodCed, 2018-09-12:
@kimchi88 you need to adjust the interfaces in the ODAS configuration file.
I suggest you use:

ssl
format = "undefined"
type = "blackhole"

sst
format = "json"
type = "terminal"

separated
type = "file"
path = "separated.raw"

post-filtered
type = "file"
path = "postfiltered.raw"

This should give you tracked source display in the terminal, while saving the processed audio.
You'll be able to import the raw PCM audio in audacity and listen to each channel (separated source).

GodCed, 2018-10-11:
Hi, I imported your raw output in audacity and it sounds allright.
I used those settings:

[image]

Did you specify 4 channels (one per tracked source)? That would be my guess.

gloriafer18, 2019-04-10:
Hi, I configured ODAS (Core and web). I was able to see the visualisation (remotely). I was working with a Matrix-voice (8 microphone array) and a Raspberry 2. After trying to configure so many times I still have a strange effect in the recorded files, when I open them by using Audacity or other software (Switch) to open RAW data, the audio sound very fast. Most of the information on internet suggest to configure the same Frequency when recording to open the file, but I can not make it work. I attached my configuration file. Might be you could tell me what is wrong with it? I also attached the audio I recorded by using ODAS.

[FilesODASMatrixVoice.zip](https://github.com/introlab/odas/files/3062178/FilesODASMatrixVoice.zip)

Any comment could be very helpful. Thanks

I want to do something similar like @chad1023 Did you succeed in doing it?

GodCed, 2019-04-10:
Using those settings in Audacity, which are the same as in my previous answer, it sounds alright.
[image]

You'll want:
- Signed 16 bit PCM
- Little-Endian
- 4 channels (one per tracked source)
- 16,000 Hz sampling rate

gloriafer18, 2019-04-11:
Thanks! sorry, I was setting 8 channels each time I open the file. But, by using 4 channels it works properly. :)

GodCed, 2019-04-11:
Glad it's solved!

closed 2019-04-11 by GodCed