introlab/odas #113: odas with respeaker 6-mic array

goldenqwerty, 2018-08-26:
tried using the respeaker 6mic array with the following [config file](https://github.com/xiongyihui/odas/blob/master/config/odaslive/respeaker_6_mic_array.cfg)

similar to issue #39 , 
there is an error:
Source hops: Cannot set sample format: Invalid argument.

tried changing the nBits to 16,24,32 but all did not work.
Is there anything wrong with the config settings?

taospartan, 2018-08-26:
What does  arecord -l  say?

goldenqwerty, 2018-08-26:
card 1: seeed8micvoicec [seeed-8mic-voicecard], device 0: bcm2835-i2s-ac10x-codec0 ac10x-codec0-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

goldenqwerty, 2018-08-26:
i guess its seeed8micvoice as i have added the:

# Seeed Voicecard
ATTR{id}=="seeed8micvoicec",ATTR{number}=="1",ENV{PULSE_PROFILE_SET}="seeed-voicecard.conf"

in 

/lib/udev/rules.d/90-pulseaudio.rules

as suggested [here](http://wiki.seeedstudio.com/ReSpeaker_6-Mic_Circular_Array_kit_for_Raspberry_Pi/)

taospartan, 2018-08-26:
Have you tried recording audio with arecord?

goldenqwerty, 2018-08-26:
Yeah arecord works. Audacity too

goldenqwerty, 2018-08-26:
arecord -Dac108 -f S32_LE -r 16000 -c 8 a.wav

taospartan, 2018-08-27:
Change this

potential: { format = "json"; interface: { type = "socket"; ip = "127.0.0.1"; port = 9001; };
To this

potential: { format = "undefined"; interface: {             type = "blackhole"; };

And

tracked: { format = "json"; interface: { type = "socket"; ip = "127.0.0.1"; port = 9000; }; };

To

tracked: { format = "undefined"; interface: { type = "blackhole";  }; };

Then let me know

goldenqwerty, 2018-08-27:
it's the same issue:
+--------------------------------------------+
|    ODAS (Open embeddeD Audition System)    |
+--------------------------------------------+
| Author:  Francois Grondin                  |
| Email:   francois.grondin2@usherbrooke.ca  |
| Website: introlab.3it.usherbrooke.ca       |
| Version: 1.0                               |
+--------------------------------------------+
| + Initializing configurations...... [Done] |
| + Initializing objects............. [Done] |
| + Launch threads................... [Done] |
| + Threads running.................. Source hops: Cannot set sample format: Invalid argument

goldenqwerty, 2018-08-27:
not sure if i'm doing the right thing:
before i run this 
(pi@raspberrypi:~/odas $ ./bin/odaslive -c ./config/respeaker_6_mic_array.cfg -v),

 i also do this  pi@raspberrypi:~/Download/odas_web-0.2-alpha $ npm start

goldenqwerty, 2018-08-27:
arecord -L gives the following (is it normal that the default is through Pulseaudio sound server?):

null
    Discard all samples (playback) or generate zero samples (capture)
default
    Playback/recording through the PulseAudio sound server
ac108
dmixer
ac101
sysdefault:CARD=seeed8micvoicec
    seeed-8mic-voicecard, 
    Default Audio Device
dmix:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct sample mixing device
dsnoop:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct sample snooping device
hw:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
    Direct hardware device without any conversions
plughw:CARD=seeed8micvoicec,DEV=0
    seeed-8mic-voicecard, 
   Hardware device with all software conversions

goldenqwerty, 2018-08-30:
@xiongyihui please give some advice on this issue where odas is giving " Source hops: Cannot set sample format: Invalid argument" using the respeaker 6 mic array

xiongyihui, 2018-08-30:
Sorry, I made a mistake, the sample bits must be 32, then it should work. I just tested it. 

A modified version which outputs message to terminal instead of socket is:
[respeaker_6_mic_array.cfg](https://github.com/introlab/odas/files/2336124/respeaker_6_mic_array.txt)

Do you use pulseaudio?

goldenqwerty, 2018-08-30:
@xiongyihui 
yea I used 32bit but mine wasnt working; 
yea the pulse audio was installed for the snowboy doa function (under step 2 of play with snowboy as suggested in this [link](http://wiki.seeedstudio.com/ReSpeaker_6-Mic_Circular_Array_kit_for_Raspberry_Pi/)
is the pulseaudio messing with the config?

xiongyihui, 2018-08-31:
Maybe uninstall pulseaudio and just read audio from alsa.

goldenqwerty, 2018-08-31:
working now, i was pointing to a wrong config file hahaha

closed 2018-08-31 by goldenqwerty

deeprobo-dev, 2022-08-03:
sudo aplay -l run this in terminal to know what is your card no and put it in odas.cfg file