introlab/odas #114: A question about kalman fliter.

huotuichang1, 2018-08-29:
what is the mean of B1,B2,B3,B4 in kalman2coherence_process? It seems not only the kalman fliter.
If there any book or article of that?I want to learn more about that.
And another question about the sss.If we can use NMF to separate the voice<,Maybe It is also a good way.

FrancoisGrondin, 2018-09-05:
Hi,
B1-4 are intermediate values used to estimate the closed form expression in the modified Kalman filter used by ODAS. While the paper describing in much details this has been accepted and the minor revision sent, I'm still waiting for the official publication. In the meantime, we can refer to my PhD thesis which covers this topic. See equations 4.38-4.41.

https://savoirs.usherbrooke.ca/bitstream/handle/11143/11621/Grondin_Francois_PhD_2017.pdf?sequence=8&isAllowed=y

huotuichang1, 2018-09-06:
Thanks.
Actually,I try to find some model like gmm to explain the kalman2coherence and the mixture2mixturee process.But It seems like that is not only the model.And I read the article about the manyears.The model is not sample to the odas.
And for more,maybe the postfliter can be better.

huotuichang1, 2018-09-10:
I see the article of you and understand the process.Thanks!

huotuichang1, 2018-09-11:
A small question.
[image]
why the (2PI)^{2/3} not the (2PI)^{1/2}?

huotuichang1, 2018-09-11:
emmm,And what the swapping means?

huotuichang1, 2018-09-12:
why we can calculate the scaleFactor as so on...I need your help.Thanks!

huotuichang1, 2018-09-12:
And in (4.49)the discrete Kronecker delta has a little mistake.