introlab/odas #115: Changing the nPairs during localization.

alexengbloom, 2018-09-07:
I am trying to change the number of pairs used for localization. It seems like ODAS is doing permutation and is using all possible pairs as possible.

If I simply try to modify the freq2freq.c (freq2freq_product_process)
and try to hard code obj->products->nPairs to a smaller number I get a Bus error; memory problem.
Changing the nPairs in the mic.c does the same.

It's quite difficult to know which part I should modify.
I tracked the problem and it seems like the error is caused by
line : thread_join(aobjs->asrc_hops_mics_object->thread); in function threads_multiple_join()

Could you help me solve the problem?

Thanks!

FrancoisGrondin, 2018-09-08:
Hi Alex,

Sorry for the late reply. You should normally change the number of microphones, and the number of pairs will be calculated automatically N = M(M-1)/2, where N is the number of pairs, and M is the number of microphones.

Out of curiosity, is there a particular reason why you'd need to change the number of pairs?

Thanks,

Francois

luc2yj, 2018-10-22:
Are there any updates on this issue? I would also be interested in this feature.

Do you (@FrancoisGrondin ) have any suggestions or ideas of how to achieve this in a simple manner? Did you (@alexengbloom ) make any progress?

alexengbloom, 2018-10-22:
@luc2yj Not yet. @FrancoisGrondin Would you have a chance to look at this issue? Or provide some advice on how to make the modification?