introlab/odas #119: Is that the Joint Probabilistic Data Association Filter?Or the Improvement of it？

huotuichang1, 2018-09-27:


FrancoisGrondin, 2018-10-10:
Not sure I do understand your question; can you be more specific?

huotuichang1, 2018-10-10:
OH，I'm sorry about that.My question is that I found the some part of M3K  seems like that is a improvement of the Joint Probabilistic Data Association Filter(in kalman filter) at that time.So I ask the question.

FrancoisGrondin, 2018-10-10:
The proposed Kalman filter uses both energy and position of potential sources to perform accurate tracking, and is a replacement of the traditional particle filter which requires more processing power. There is a matching between observations and tracked sources that is done to perform multisource tracking. Is this what you are asking? Thanks

huotuichang1, 2018-10-11:
yeah,I get it! Thank you very much.
I was reading your paper in recent two month,it is very valuable! 
And for more,I have a small question:
[image]

Is the (2PI)^{2/3} should be the (2PI)^{-3/2}?